PiSupply/PiJuice #172: Unusually high current

Reefkeeper, 2018-07-17:
When I watch the PiJuice-CLI -> HAT status the GPIO power input seems rather high.
I says current is always higher than 1.5 A, sometimes up to 2.2 A
My Raspberry Pi 3B+ has a load of below 5 %. If I start "htop" this is the heaviest load.
Any ideas?
[image]

ChristopherRush, 2018-07-17:
What sort of power supply are you using, is it regulated?

tvoverbeek, 2018-07-17:
Looking at your status display, you are running on battery only, no power connected to either RPi or PiJuice.
I can easily get this amount of current on a 3B+ when running Chromium and displaying a dynamic website (many moving ads, etc.).

Reefkeeper, 2018-07-17:
@ChristoperRush:The power supply is regulated (for Raspberry Pi) 5.1 V @ 2.5 A.  It has been used for months for other RPis without problems, so I suppose this is not the culprit.
@tvoverbeek: You're right, when the power is connected to the RPi the current drops to 0.4 - 0.8 A which is reasonable. 
One thought was that the current with 3.7 V (battery mode) must be higher than with the 5 V power supply, but this would only count 1.35 times higher. Still puzzling.

Reefkeeper, 2018-07-17:
When built into the small PiJuice case the battery gets very hot (cannot keep fingers on the surface).

closed 2019-09-15 by ChristopherRush