PiSupply/PiJuice #174: Shutdown state - turning off 5VDC on GPIO pins

dannaughton, 2018-07-23:
I have a PIJuice with RPI 3 Version B with the pogo pin install.  Everything seems to be working correctly.

When the PIjuice shutdown down the RPI after SW1 button press, the 5VDC on the GPIO stays at 5 Volts.  Is that the correct behavior.  If so, I guess your not supposed to connect anything to that supply, or your battery will drain even with the PI shutoff.  Is that correct?

Background info:
I'm trying to build a RC car with the rpi, and I've burrned through two of these battery baby sitter boards from SparkFun.  This PIJuice was a lot more money, but seems to be working well.  Although I can't find a good reference design for the giant battery I'd like to use for the motors.  I don't think connecting the motors the 5VDC from the pi is a good idea.  Any advice from a EE would be appreciated.  Thanks.

tvoverbeek, 2018-07-23:
@dannaughton In the Pijuice settings app (pijuice_gui for the desktop, pijuice_cli.py for the terminal/remote ssh) you can change the setting for the SW1 long press from the default SYS_FUNC_HALT to SYS_FUNC_HALT_POW_OFF.
This will remove the 5V power to the RPi 60 seconds after the halt command is set by the PiJuice.
The RPi red led will then go off.
For the motors use a separate supply.

codejedi365, 2018-09-02:
@tvoverbeek & @dannaughton Thanks for this post! It solved my issue too.  I didn't realize the default was SYS_FUNC_HALT.  When I made the change, the power turned off exactly in time.  

My situation was similar but in my project I am using a 3.5in LCD touch screen for the RaspberryPi and the backlight would not cut off after SW1 press even though the OS was shutdown.  Now with this post, all is well.  Thanks again!

tvoverbeek, 2018-09-02:
Closing the issue. if you do not agree, please reopen it and indicate why.

closed 2018-09-02 by tvoverbeek