PiSupply/PiJuice #175: Communication-error

kf4jbz, 2018-07-23:
Have 2 pijuices. First one worked out of the box. Updated firmware worked ok. Removed pijuice from pi for height testing of standoff’s and stacking headers to mount in smarti pi case. Removed all power sources before testing. Reinstalled pijuice got communication-error in GUI Battery, GPIO, fault.  Second pijuice same GUI communication-errors out of the box. Using same pi2. Tried reboot, software/firmware update, and reinstalls, updates, upgrades, boot loader.

tvoverbeek, 2018-07-24:
Are you sure you mounted the PiJuice correctly on the 40-pin RPi header? I myself have a few times misplaced the PiJuice (one pin off, or even missing a complete row) which obviously resulted in communication errors (=no I2C communication). Does `i2cdetect -y 1` report the two addresses of the PiJuice (0x14 and the RTC at 0x68)?

kf4jbz, 2018-07-24:
I am sure that both PiJuice#1 and #2 is mounted correctly.  Tried i2cdetect -y 1. Will try to send a screen shot and copy of emails from Chris at PiSupply.
Thanks

kf4jbz, 2018-07-24:
Should I try another rpi2?  Any idea where the communication_error is coming from? i.e.,between the pijuice and the rpi, between software and pijuice, or other? The pijuice  green/blue/red leds are working.

kf4jbz, 2018-07-24:
Screen shots 4 Communication_error
[image]
[image]
[image]
[image]

tvoverbeek, 2018-07-24:
Since i2cdetect does not show anything there is obviously no I2C communication between the RPi and the PiJuice. PiJuice software is not involved with i2cdetect.
Here is what to expect from `i2cdetect -y 1` with the PiJuice mounted:
```
pi@rpi-stretch-full:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                   
```
Have you checked if I2C is enabled after you reinstalled Raspbian?
If I remember correctly I2C is not enabled on a fresh (re)install.
On the desktop: Preferences->Raspberry Pi Configuration->Interfaces

kf4jbz, 2018-07-24:
Changed the interface tab I2C to enable, checked I2C again no change no 14 or 68, checked pijuice settings same problem-communication_error.  Any other changes to try?

shawaj, 2018-07-24:
Have you rebooted after turning i2c on?

kf4jbz, 2018-07-25:
New problem-rebooted after turning on I2C-get the rpi welcome, systems-host named.service in lower left corner and a blinking cursor. Does not complete boot up.

kf4jbz, 2018-07-25:
Third try worked system booted. Yes rebooted after changing I2C to enable and no change to i2cdetect -y 1 no 14 or 68

kf4jbz, 2018-07-25:
And yes clicked ok before reboot

tvoverbeek, 2018-07-25:
Starts to smell like a corrupt SD card and/or a bad power supply.
I am out of suggestions now.

kf4jbz, 2018-07-25:
Have 2 SD cards (16 GB Adafruit NOOBS and 8 GB Mike Richards Ham Radio NOOBS card from UK) Been trying everything suggested with one card at a time and getting same result. Process of elimination leaves the pi2. Other than the pijuice comms issue everything works. Running mains 5V 2.5 amp, rpi2, WiFi, bluetooth, 7 inch touch screen.

kf4jbz, 2018-07-25:
Thanks for your help tvoverbeek

kf4jbz, 2018-07-26:
Got another rpi2, pijuice now works. Chris at Pisupply close my ticket. The international shipping USA to UK cost more than second rpi2 at Amazon. Don’t know what is wrong with first rpi2, but everything on it works except with the the pijuice.

ChristopherRush, 2018-07-27:
@kf4jbz If you open a Github issue then your support ticket will get closed.

kf4jbz, 2018-07-27:
Questions 
From the pijuice hat config battery menu-would you recommend a battery temp sensor hat compatible with pijuice hat?
Pogo pin-function?

closed 2018-07-29 by kf4jbz

shawaj, 2018-07-29:
@kf4jbz - pogo pin is to restart pi when power plugged into the pi micro USB

There is already a temp sensor in the battery and on the board

lbabaly, 2018-08-05:
Hello,

I had the very same issue, and in the end I had switch to another raspberrypi 3 B+.
Which RPI version did you have this issue?