PiSupply/PiJuice #176: Powering on a pi after manual shutdown + power loss

cmsimike, 2018-07-24:
Hello,

Is there a way to power down the raspberry pi manually (meaning I personally run ```sudo shutdown -P now```), disconnect the power, and have the pijuice boot up the pi when the power is reconnected?

So far my tests don't seem like that is possible but I'm hoping I am missing something.

I have confirmed that if the pijuice shuts down the pi itself, it can boot it up, but seemingly doesn't do this if I shut down the pi.

tvoverbeek, 2018-07-24:
When the pijuice boots the Pi it disables the wakeup-on-charge, which is why the RPi does not reboot when you reapply the power. When the pijuice shuts down the Rpi it enables the wake-up-on-charge just before the shutdown command is sent.
You can do the same with a small python script to shut down the RPi:
```python
import pijuice
import os
pj=pijuice.PiJuice(1, 0x14)
pj.power.SetWakeUpOnCharge(0)
os.system("sudo halt")
```
When you then remove the power from the pijuice and later apply it again, the RPi should boot.

cmsimike, 2018-07-25:
@tvoverbeek Worked perfectly! This actually worked out in my favor. I currently have "Wake up on charge" set to 0, so on my first test, while the power was connected, I ran the python script and it shut down my pi, but instantly turned it on. 

For my specific case, a script will be triggered after 5 mins of no power, so the this works, as the pi shuts down and stays down without power. Then on application of power, it will boot up.

Thanks again!

closed 2018-07-25 by cmsimike