PiSupply/PiJuice #177: Raspberry brutally shuting down when on battery

sakura-lain, 2018-07-25:
Hi there,

For maybe about a week or two, my RPi 3B+ seems unable to stay on more than a few minutes when on battery with the PiJuice, even when it is fully charged. It always ends to brutally shutdown, and when I try to turn it on again it cannot end to start the system. This morning for instance I could only boot once, it stayed on for one minute or two and then turned off without any warning and from then when I try to boot it it turns off just few seconds after the procedure was launched. So I can say it is going worse and worse.

I made several tests, tried to remove all unecessary items (like the touch screen - though even with it it worked perfectly until then), tried to remove the battery for a few minutes, tried to update the firmware, tried to remove pijuice-gui... Nothing seems to solve the problem, which occurs both with a standard Raspbian and with Retropie, each of them being installed on two different SD-cards. My pijuice-gui and both systems are also up-to-date.

The RPi works perfectly when it is plugged. Must I conclude that my battery is dead ? It doesn't seem so however (my voltmeter indicated that it delivered 3.98 V when I measured it this morning).

Thanks for any help with this issue.

sakura-lain, 2018-07-25:
Hi again,

Something new just occurred : if I plug the RPi through the PiJuice micro-USB my RPi cannot start, it keeps on rebooting. It still works fine when plugged directly on the RPi micro-USB, however I had to remove the Pi in order to boot after a shutdown since - another new issue - pressing the SW1 button is of no effect.

Another thing I'm testing : it seems that when the Pi's off after having been shutdown, the screen now remains enlightened when on battery, which could prove that there is no problem with the battery.

tvoverbeek, 2018-07-25:
While powering the RPi (to prevent it from shutting down) check the setting of 'minimum charge' on the 'System Task' tab. If it is enabled and the threshold setting too high you get this 'immediate shutdown' effect especially with higher loads (3B+ + touch screen is a considerable load). With higher loads the battery charge measurement indicates lower battery charge level than actual (due to the resistance of the internal battery protection circuitry of the BP7X).

sakura-lain, 2018-07-25:
Yes I had this indicator problem before but with no consequence on the actual performance until recently. I just checked the system task and everything is disabled. Maybe should I set something? If so, what do you advise me?

Thanks again.

sakura-lain, 2018-07-26:
Hi again,

I tested several things with task manager with no success. Also checked that GPIO input and charging were enabled as indicated [here](https://github.com/PiSupply/PiJuice/issues/113), try to charge the PiJuice separately, cleaned up the connectors...  No idea what else could be done or checked. I am pretty sure it is a problem of charge not being correctly detected, but how and why, that is the question. No idea if it is a hardware or software problem either. It is quite strange since until two weeks ago everything worked fine and I can't remember having changed something inbetween.

sakura-lain, 2018-07-29:
So I tried with a Pi Zero with Raspbian, it seems to work better even if it crashed once or twice when plugging a mouse. Didn't try with my touch screen yet, I'm only testing it on my TV screen for now. The battery level indicators still seems inaccurate but anyway, well, the Pi stays on. I am now making a attempt to measure for how long. 

The main problem I encounter for now is that the PiJuice cannot send the Power on signal when SW1 is pushed when it is unplugged, I can only hear a worrying sizzling, whereas it normally works when it is plugged.

sakura-lain, 2018-07-29:
I ran a test consisting in installing Raspbian through Noobs to know if it could change anything, but the result is the same. :-(

So to summarize : 
- the PiJuice seems to work rather good with my Pi Zero, although I am still skeptical about its autonomy and despite a few bugs encountered.
- The problem arises with my Pi3B+ : the Pi is unable to end the start sequence and crashes when on battery or endlessly reboots when plugged through the PiJuice. It crashes when unplugged.
- In both cases, everything runs well when the Pi is plugged through its own USB powering source.
- Making minimal installations and/or removing all unnecessary peripherals doesn't change anything to this behaviour.
- Disabling everything in 'System Task' tab or trying different settings didn't change anything either.
- The problem is going worse, since I am now unable to make my Pi3B+ work with the PiJuice, whereas it used to work fine until approximately two weeks ago and could still last week at least start, even for a few minutes.
- I can't determine what may have change between the time it used to work and now.

I'am stuck there... Any other ideas ?

Thanks again!

ko4mz, 2018-09-07:
Hi, The exact same issue has happened to both my PiJuice Hats as sakura-lain.

First, the battery dies within about 1.5 minutes when the micro USB PS is disconnected or turned off.
Second, after 48 hours of running/rebooting of the first PiJuice, it died completely and I had to remove it from the Pi 3B+.
The second PiJuice, just installed today, is doing the same thing with the battery dying in less than 2 minutes.  Only time will tell if it will stop working altogether.  I returned the first Pi and have received a full refund - purchased on Amazon, credited by Amazon.

I also tweeted about the issue to #PiSupply and was asked to resend the issue to #GitHub and haven't received a reply.  I see that there are other threads on the battery issue so I certainly hope this becomes high priority.  I purchased the PiJuice for the RTC and especially for the battery backup feature, and later plan to go with a flexible/portable solar panel, but only if this issue is resolved (and safe).  There is about  a thousand other folks with the same mindset following on facebook/youtube concentrating on emergency communication with ultra portable power.

Thanks for listening.

tvoverbeek, 2018-09-07:
@ko4mz Maybe a red herring, but two things to check:
- Has the plastic tab between the battery and the battery contacts been removed?
- In the GUI/CLI check if charging is enabled. On my own PiJuice I have seen a case where charging was disabled.

sakura-lain, 2018-10-01:
Hi,

Finally got a new PiJuice thanks to warranty, it solved the problem. :-)

ko4mz, 2018-10-01:
My second new PiJuice HAT still has the same issue. Less than 2 minutes, the pi is dead.

ko4mz, 2018-10-02:
> It seems like you likely have it set up incorrectly, or some other issue / clash because they are all tested out of the factory so this is not a likely scenario.
>
> Where did you send the previous one back to? I'd like to try and get my hands on it to test.

Amazon.  It was purchased via Chicago engineering folks. 

There's not much to set up the HAT, and charging was enabled.  It should have lasted for about 4 hours.

Thanks. Eric.

sakura-lain, 2019-04-22:
Hi ! 

I having the problem again on my second PiJuice. I found another guy encountering the same problem on a French forum : https://www.framboise314.fr/pijuice-une-alimentation-ininterruptible-ups-pour-votre-raspberry-pi/#comment-35178

Both of my PiJuices come from Kubii, France. I sent back the first one so maybe you can get it and try it.

Symptoms are the same as my first issue :
- The problem appears after a few weeks or months of use
- My Pi 3B+ powers off or interrupt its start-up when on battery, no matter which version of Raspbian I use and including on a lite fresh install whithout the PiJuice software installed.
- My PiZero seems to correctly work on the contrary, although I have strange readings : the battery is read to be full when the Pi or the PiJuice is plugged in but when unplugged the remaining battery level seems to be correctly displayed, even if I have to wait for it between 30 sec-1min. Otherwise I have irregular and even negative readings on GPIO input (as if the battery were discharging or incorrectly charging) when plugged through the Pi (same on the Pi 3B+)

After having tried to reset it, to change a few settings, I tried a new approach : I fully discharged the battery with the PiZero, then fully recharged it (plugged PiJuice) then watched what happens :
- At first I thought my problem was solved since my fresh install of a lite Raspbian seemed to start and run correctly.
- Then I tried with my up-to-date running Raspbian (which I use as a Wifi router) and the problem immedialtely came again.
- Coming back to my lite Raspbian, I can watch that it now reboots three times before powering off.

This is really weird and disappointing, even if I feel less lonely to see that other people encounter this.

cjdcordeiro, 2019-10-18:
same rebooting issue here. This is not closed

ko4mz, 2019-10-18:
Yes, this is still very much an issue and the GitHub folks will instantly advise that the issue was your problem without even investigating further.  See response to my initial post.  This happened with two PiJuice hats.  I am done with Pi-anything.

shawaj, 2019-10-18:
Will leave this issue closed as I believe #397 and #371 are similar