PiSupply/PiJuice #178: Automatic power on (without S1)

insideoutgrp, 2018-07-26:
I'd like to power my RPi ideally through the Juice micro USB (gives more flexibility with power on, use of RTC etc) but need it to come on without using S1.
I've tried using the % charge system tag, but this only works if the Pi has been shut down (e.g. using the power off system halt trigger when power is removed)
Ideally I need something that powers it on under all circumstances, as the project I have in mind involves having these units in inaccessible places i.e. its possible they would run the battery flat on the juice, RTC would fail and therefore not power on using any timer etc. 
I think I read in a previous issue that in firmware 1.0 they used to default to power on - is this possible?
Thanks

tvoverbeek, 2018-07-27:
See my comments on Issue #176
Your best bet is to use the wakeup-on-charge, but you have to enable it when the Pi is running with
the SetWakeUpOnCharge() function from the pijuice module.
Probably to do this immediately after boot e.g. in /etc/rc.local.
As explained in issue #176 the PiJuice firmware disables the wakeup-on-charge when it wakes up the Pi. It works when you use the power of system halt trigger because the system task which does the 'sudo halt' enables the wake-up-on-charge just before doing the 'sudo halt'.

fedster777, 2018-10-29:
I've been reading issue #176 and this one but I still have a problem with wake on charge...
I use pijuice system_events to trigger a "SYS_FUNC_HALT_POW_OFF" on "no power", this works fine.
I configured a system task to trigger a "wakeup on charge" set at "0", that works but only within 60 seconds of system halt (as long as red led on rpi is still on, that is power still available). If I wait more than a minute and then try to wake up rpi connecting power to the micro USB on pijuice, it doesn't wake up... any ideas why?

fedster777, 2018-10-30:
Here is the  "pijuice_config.JSON":

{
  "system_events": {
    "no_power": {
      "function": "SYS_FUNC_HALT_POW_OFF",
      "enabled": true
    }
  },
  "system_task": {
    "enabled": true,
    "wakeup_on_charge": {
      "enabled": true,
      "trigger_level": "0"
    }
  }
}

ChristopherRush, 2018-10-30:
Have you updated the Firmware?

tvoverbeek, 2018-10-30:
1.3.3 is the debian package version, not the firmware version.
You will find the firmware version on the firmware tab in the gui or the Firmware entry in the cli.
Latest released is 1.2 (which is **not** the version as delivered in the PiJuice, you need to update it to 1.2) 1.2 contains fixes wrt wakeup.

I tested on my PiJuice with your setup and it works for me. Pi shuts down when power is removed and boots when power is reapplied, both with the Pi power led still on and also after it goes off.

closed 2019-09-15 by ChristopherRush