PiSupply/PiJuice #180: Weird battery percentage PiJuice CLI Status

jordiblanchcarles, 2018-07-28:
I've purchased a new PiJuice, and installed the pijuice-base package because I'm using it with a headless RPi 3B+. I've also updated the firmware (through PiJuice CLI) to v1.2. After more than 16h charging, the board is still indicating that the battery is charging (blue led blinking) and the CLI Status indicates 99% charge. But some seconds after disconnecting the power supply, the CLI indicates 17% charge, and after a couple of minutes the battery has fallen down below 5% (which I've setup as the event to shut down the Pi). I don't think my PiJuice is working properly, and of course, the BP7X battery is not lasting for the time that PiJuice is indicating at their website.
Before asking for a replacement, I'd like to know if there's anyone here that can tell me if there's some kind of problem with my hardware (and battery) or there's any kind of software configuration that solves the problem, because it is a brand new device.

Thank you!

Reefkeeper, 2018-07-30:
Same with my PiJuice: rapid drop of the percentage after beginning the battery mode. But then it still takes about 1 hr 40 min before the Raspberry reaches 1 % and really shuts down. The discharge curve is - how can I put it - **very** non-linear.
Hope this will be fixed sometime!?

shawaj, 2018-07-30:
This is documented on GitHub already. It's because there is built in protection in the battery and also on our board which confuses the battery fuel gauge chip.

We may be able to fix this in software in the future but it's not a priority

jordiblanchcarles, 2018-07-30:
Hello, thank you for answering, but IMHO, having a minimal accuracy of the power left on a UPS system sounds like a kind of "priority". Today I've done a test, and after a lot of hours charging, after unplugging the power supply, the battery percentage has fallen down below 50% almost immediately, and after 2 minutes i was below 15%, and in about 5 minutes it was below 5%, and then it started falling down very slowly and after more or less 1h i was at 2%. At least to me, this is not very far from just guessing the power left on the device, because if I want the RPi to be up as much as possible but having enough time to shut down properly, with the current hardware/firmware/software combination all I can do is to set up a lower limit of 1% battery left to shut down (otherwise the RPi would shut down in some minutes and PiJuice is supposed to have enough power to last 3/4 hours depending on the power needs), and setting the lower percentage to 1% doesn't seem to me a very good choice if I expect the system to always have enough time to shut down properly. Besides, this 1% percentage is a test/error value, and it doesn't give me any kind of confidence that it will continue being a "safe" value to trigger the shut down process for a long time (months/years). I can only "cross my fingers" and expect that it will.

Sorry for my long answer, but I really think that some accuracy in knowing the remaining power in a UPS system IT'S A MUST and I was trying to explain why (from my point of view).

continuum-kevin, 2018-11-07:
I assume that since firmware is still at v1.2, this has not been fixed yet. Any update on this? If I can't get an accurate software reading on battery percentage, PiJuice is not actually very usable.

ChristopherRush, 2018-11-07:
We are currently testing the Beta Firmware v1.3 internally.

shawaj, 2018-12-11:
Check this for the beta firmware - https://github.com/PiSupply/PiJuice/issues/249

ChristopherRush, 2019-09-15:
1.3 Firmware introduced in Software v1.5 - f571b4384aea02c75073204db5213538a9c02fbf

closed 2019-09-15 by ChristopherRush