PiSupply/PiJuice #181: how to restart inactive service?

daviddnsw, 2018-07-29:
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; disabled; vendor preset:
   Active: inactive (dead)

tvoverbeek, 2018-07-29:
Try the following:
```
sudo systemctl enable pijuice.service
sudo systemctl start pijuice.service
```
After a normal install of pijuice-base the service should be enabled.
However it can still die due to errors (e.g. no PiJuice connected at system boot).
After a while check again with
```
systemctl status pijuice.service
```
Maybe the output gives some hints why it died (in case it dies again after the start command)

daviddnsw, 2018-08-04:
thanks! that fixed it and all sees stable after several reboots

closed 2018-08-04 by daviddnsw

shawaj, 2018-08-04:
@ChristopherRush any idea if this is already documented somewhere?