PiSupply/PiJuice #184: LED1 lights up every 16 minutes in deep sleep mode

pat160, 2018-08-08:
Dear,

when the PiJuice is in deep sleep mode (LED1 flashing faintly green every +/- 4 seconds), I also noticed that LED1 lights up very bright (green color) during approx. 3 seconds, every 16 minutes.
Why is that ?

Regards,
Patrick.

ChristopherRush, 2018-08-09:
It is just to let you know what the current battery status is i.e. charge level

jdotbdot, 2018-11-03:
Can you explain the difference between the frequent faint flash and the infrequent blinding one?
Is it possible to disable either or both of these?

tvoverbeek, 2018-11-05:
@jdotbdot My guess is that a timer in the firmware is overflowing causing the MCU to leave the deep sleep for a while and since the clock is running slower it results in a longer flash of the load indication led (D1).
If you want to disable the LEDs just set them to 'NO FUNCTION' in the gui/cli. However this will also disable the LEDs for normal PiJuice operations.

closed 2019-09-15 by ChristopherRush