PiSupply/PiJuice #185: Problems with RTC and System Tasks

richardquint, 2018-08-10:
I'm having problems with my PiJuice that I am unable to successfully troubleshoot myself.  I have tried resetting the PiJuice (first pulling the battery and then after replacing it pressing SW1 and SW2 simultaneously for over 20 seconds until LED2 blinked and finally turned green) and I have removed and reinstalled both the pijuice-base and pijuice-gui packages.

The PiJuice's RTC doesn't advance after a full shutdown of Pi and PiJuice.  After the Pi boots there is a delay of about 90 seconds before the Pi gets the correct time from its internet connection during which the time displayed is the same as when the Pi was powered down (I also checked this by writing the time that rc.local runs to a log file).  Also the battery seems to last only 45 minutes after being fully charged.  I am using a Pi3B+ with a new Adafruit 2.5A power source, nothing else attached to the Pi and just letting the Pi sit with the PiJuice GUI running.  I have successfully set SW2 to do a HARD_FUNC_POWER_OFF and SW3 to do a HARD_FUNC_POWER_ON with a single press and also have a python script to shut everything down.  However I can not get a System Task (Minimum charge [10%]) + System Events (Low charge) to run a USER_FUNC1 (which is to run the aforementioned python shutdown script) to work.  I have checked that all permissions are set on the scripts and the log files are writeable by everyone.

tvoverbeek, 2018-08-11:
The RTC is advancing, but it is never read by Raspbian at boot to copy its time to the system clock.
When the system clock is set from the internet, Raspbian also copies the system time to the RTC.
What do you get if you do a `sudo hwclock -r` just after boot before the system clock is synchronized from the internet by systemd-timesyncd?

Have you tested your USER_FUNC1 by assigning it to e.g. a single press of SW3?
That way you know your function is OK.

Am on travel now and have no access to my RPis

richardquint, 2018-08-12:
I was using a script in rc.local that called the date command to get the time.  Also running sudo hwclock -r in rc.local confirmed your explanation. 

I have changed the assignment for SW2 to be my script for shutdown and it works exactly as expected. I can see it even without logging the time because I have LED2 blink a few times in before the final halt command is run.  I'll continue to experiment, but it seems that the percentage setting (I set it at 10%) for low charge  is ignored and it isn't run until the charge is down to 0%.  Is there an easy way to include a reading of the charge level when the shutdown script is run?  I'd also like to log the time when power to the PiJuice microUSB input is turned off and include the charge level there too.  That way I can have a record of how long the battery lasts.  

I have purchased a second PiJuice (an outfit called Copperhill Technology here in the US had a great sale), and the length of time for the battery (same model on both units) to fully discharge is very different.  I think I may try swapping batteries and check if it's the battery or the PiJuice causing the short discharge time,

richardquint, 2018-08-20:
I'll close this.  See issue #191 for any followup.

closed 2018-08-20 by richardquint