PiSupply/PiJuice #186: RealTime clock documentation

oh8stn, 2018-08-12:
Hello
where can I find detailed documentation on the real time clock?
I have an off grid, no internet project requring a time accuracy no greater than +/-2 seconds. Other than its existence, I have not found any documentation about how to use teh real-time clock to keep the pi "on time" when away from the internet.
Some questions
How to configure the real time clock?
How does the Rpi get the accurate time from the real-time clock?
Can I force the PiJuice to automatically update the time on the RPi?
How accurate is the real-time clock?
Are there any code or configuration examples?
---

On a related topic-
The documentation is scattered across the globe. Can we please have one PDF document with all documention about the PiJuice, its functions, hardware, switches, EVERYTHING all on your website rather than forcing newcomers to search the world for documents? 

thanks 
Julian OH8STN

tvoverbeek, 2018-08-12:
@oh8stn If you are using the PiJuice HAT ID-EEPROM at its default 0x50 address, you can use the standard Linux `hwclock` utility to read and set the time of the RTC.
The RTC is simulated by the PiJuice MCU so its accuracy is determined by the accuracy of the clock generator in the MCU. To keep the RTC running there must be some battery capacity left. When there is no other load the PiJuice MCU goes into a low power mode, but the simulated RTC keeps running.
Short term accuray of +/- 2 seconds should be no problem. I do not know about long-term accuracy (weeks, months continuous use).
Since you are considering a stand-alone no internet application (Somewhere near Oulu, FInland?) here is how you can approach it.
Assumption: Raspbian-Lite, no desktop
- You only need the pijuice-base package.
- Without internet, after first boot set  the date and time using the Linux `date` command. This sets the system clock
- Copy the date and time to the RTC: `sudo hwclock --systohc`
- On subsequent reboots the system clock has to be initialised from the RTC: `sudo hwclock --hcto sys`. You can do this with e.g. a cron job using the @reboot item or put it in /etc/rc.local. Make sure you test for the availability of I2C since cron is started before the I2C modules are loaded. 
You probably want to remove the fake-hwclock package just to be sure.

See also my example Python program for a slightly different scenario here: https://github.com/PiSupply/PiJuice/issues/91#issuecomment-381324352

For documentation the best place are the general, software and hardware README's on github.
Final resource is the actual code (which is what I had to use to set the alarm from Python in the example).

Hope this puts you in the right direction

ve3obo, 2018-09-21:
Thanks for asking this, Julian. 
I am a newbie to raspberry pi and pijuice. I was able to get the time reset with "sudo hwclock -s" a couple of times but then it stopped working. I have no idea why. The pijuice does hold the correct UTC time but I cannot figure out how to update the main RPi clock so that I can run Ft8 and ft8call. 
Btw, I have used ft8 and ft8call to activate summits on SOTA. Works great! I have been lucky that I have had internet service on the summits and could set the time through my phone as a hot spot. 
I wish there was some way to just manually set the time on RPi as we only need +-2 seconds. My watch can do that. 
I will try the suggestions above but I'm not hopeful. Being a newbie can be very frustrating trying to decode what instructions mean. 
Bob ve3obo

ve3obo, 2018-09-21:
Just as I feared, nothing works. When I type I "sudo hwclock -s" it comes back with "cannot access the hardware clock via any known method" the same is true for other arguments such as -r. I am stumped, especially since it worked a couple of times in the beginning. I had to do lots of hard shot downs because I didnt know about the "system task enable" button. That cost a few hours. Seems like default would be "enabled" not disabled. Anyway, I missed that instruction. 
So now I guess there is something wrong with hwclock which is a RPi problem but I hesitate to reload the Raspbian stretch software because it took soooo long to get it working with the ham radio software. 
Help!
Bob ve3obo

tvoverbeek, 2018-09-21:
@ve3obo Obviously the real-time clock os module is ot loaded.
Check the output of `lsmod` in a terminal.
If the line `rtc_ds1307             24576  1` is missing the clock module is not loaded.
If the module is not there check the output from `i2cdetect -y 1`.
My output is:
```
pi@rpi-stretch-full:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@rpi-stretch-full:~ $ 
```
I you see 'UU' at position 68 it means that address is handled by the OS (i.e. the clock module).
If the module is not loaded you should see `68` instead of `UU`.
Meaning the i2c address of the clock of the PiJuice is recognized.
Any other equipment (ham radio interface) using the i2c address 0x68?
Let me know what you find and we can take it from there.

ve3obo, 2018-09-21:
First of all, thanks. I can follow your instructions easily. 
I get 68 where you get UU.  Otherwise it looks the same with the 14 in the right place, no other entries.
Bob ve3obo

tvoverbeek, 2018-09-21:
@ve3obo OK, so the module is not loaded, which explains why the hwclock command is not working.
To get hte module loaded there are several ways of doing it:
1) Make sure the id-eeprom address is at 0x50 (not 0x52).
See the attached screenshot
[image]
Then reboot.
The module should be loaded now, i2cdetect should show 'UU' now and the hwclock command should be working.

2) Force loading the module by adding the following line to /boot/config.txt:
`dtoverlay=i2c-rtc,ds1339`
and reboot

oh8stn, 2018-09-21:
@tvoverbeek thanks for the reply. If it wasn't for the other person who tagged me in a post, I would never have known you replied to me.
I have a little bit more information for my setup in case it makes a difference. I'm using a GPS with chrony to get time. I decided to rely on the GPS because I couldn't validate if the RTC was working correctly when I initially posted. Actually that solution is working perfectly, but I don't want to rely on it. The RTC on the pijuice is the backup, in case the GPS fails for some reason.
[image]

I'm running full Raspbian Stretch headless, which I connect to and control using an Android tablet over Wi-Fi using VNC. The setup is used for off-grid digital comms in the field.

 [image]

The only thing left for me to do ( I believe) is figure out the way to occasionally & automatically update the RTC, from the GPS.

Finally thanks for all the information and helping to get us squared away. 

73
Julian oh8stn 
"If it was so obvious, the person wouldn't be asking for help".

oh8stn, 2018-09-21:
@ve3obo no problem Bob and don't give up. The only problem with this Hardware is the documentation. The people here seem really willing to help get us sorted.
73
Julian oh8stn

oh8stn, 2018-09-21:
@tvoverbeek thanks for the info I'm following along very closely.

ve3obo, 2018-09-21:
Yeah!!! I had to use the forced install by adding the line to config.txt. it took me a few tries to get the line error free with nano text editor. I noticed that I had added the line before but with a different four digits at the end of the line. The blog I got it from suggested 3231, not the 1339 that you suggested. I also noticed that I used a "." Instead of a "," .  That took a while for me to find as my eyes are less than perfect. 
I am very grateful for your advice. 
Btw, at first I thought it wasnt working because it took about a minute to change the system time but I knew things were getting better when I did not get an error message after entering sudo hwclock -s.  
Thanks again. Now, off to the summits I go. 
Bob ve3obo

ve3obo, 2018-09-21:
For Julian
I am a fan of yours and follow your videos. Love them. Very professional. I too have an android tablet but I have no idea how to connect wirelessly to the RPi except at home using my home wifi. In the field a use a router and ethernet cables also using vnc. I use the ft817 and the  minipro sc. I have ordered the gps dongle. For now, I am good to go. Your trailer idea will not work here as the terrain is much too rough and hilly. So weight is a huge issue. So far I have only done day hikes too summits. I do many multiday trips in the wilderness but the weight of my little station is daunting. I am wondering if I can find a lightweight qrp kit that i can set up for 7074 mhz and 7078 mhz. I would also like to get rid of the modem and the two ethernet cables. Every gram counts. 
For now i will try to stay with day trips for SOTA work. 
You are an inspiration, Julian. Hope to meet you on the air some day. 
Btw, i live in Peterborough, Ontario, Canada. About 100 km NE of Toronto. I am a retired physics teacher that loves hamming and hiking and canoe tripping. 
Bob ve3obo

ChristopherRush, 2019-09-15:
Now documented here - https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-rtc

closed 2019-09-15 by ChristopherRush

oh8stn, 2019-10-10:
Excellent,  thank you.Julian