PiSupply/PiJuice #187: inconsistent pijuice.pid addressed in _cli/_gui/_sys.py

leeee-me, 2018-08-13:
For some reason, I am working on a RPi3B with U-Boot arm64, Linux kernel 4.19 arm64, Ubuntu 18.04 aarch64 distro. And my gadget is headless which means no console UART and HDMI.

In that case, I need to rebuild the whole *light* edition of debian from scratch of source code and, I need to rely on cli version (and some py script I wrote myself). Guess what I found:
<pre>
:~/PiJuice/Software/Source/src$ grep -r pijuice.pid *
pijuice_cli.py:PID_FILE = '/var/lib/pijuice/pijuice.pid'
pijuice_gui.py:PID_FILE = '/var/run/pijuice.pid'
pijuice_sys.py:PID_FILE = '/var/run/pijuice.pid'
</pre>

Well well

tvoverbeek, 2018-08-13:
@leeee-me Thanks for catching this.
You obviously use the github version, which is ahead of the officially released version (1.3.3 from 2018-05-18). The released version does not use the pijuice_sys PID_FILE (yet).
The file is used by the gui (and the cli in the next released version) to signal the service (pijuice_sys.py) to reread its configuration file after a change.
For now just change the PID_FILE definition in pijuice_cli.py to `PID_FILE = '/var/run/pijuice.pid'`
Will update the github version.

tvoverbeek, 2018-08-13:
Github updated. Closing the issue

closed 2018-08-13 by tvoverbeek