PiSupply/PiJuice #188: [Wish/To-Do] Charging/discharging scheme to prolong battery lifetime by Sys Task & Events

leeee-me, 2018-08-14:
Suggest to add a test python script to enable/disable battery charging circuit. (maybe by different battery profile?)

Currently, I am experimenting by:
<pre>
#!/usr/bin/python
# pijuice-scan.py, can be used by rrdtool update/graph

import calendar
import time
from pijuice import PiJuice

pijuice = PiJuice(1,0x14)

status = pijuice.status.GetStatus()
status = status['data'] if status['error'] == 'NO_ERROR' else status['error']

charge = pijuice.status.GetChargeLevel()
charge = charge['data'] if charge['error'] == 'NO_ERROR' else charge['error']
fault =  pijuice.status.GetFaultStatus()
fault = fault['data'] if fault['error'] == 'NO_ERROR' else fault['error']
temp =  pijuice.status.GetBatteryTemperature()
temp = temp['data'] if temp['error'] == 'NO_ERROR' else temp['error']
vbat = pijuice.status.GetBatteryVoltage()
vbat = vbat['data'] if vbat['error'] == 'NO_ERROR' else vbat['error']
ibat = pijuice.status.GetBatteryCurrent()
ibat = ibat['data'] if ibat['error'] == 'NO_ERROR' else ibat['error']
vio =  pijuice.status.GetIoVoltage()
vio = vio['data'] if vio['error'] == 'NO_ERROR' else vio['error']
iio = pijuice.status.GetIoCurrent()
iio = iio['data'] if iio['error'] == 'NO_ERROR' else iio['error']

#print status
#print fault
epoch = calendar.timegm(time.gmtime())
#print epoch, ' ChargeLevel=', charge,'%,', 'TempBatt=', temp, 'C, VolBatt=',vbat, 'mV, CurrBat=',ibat, 'mA, VolIO=',vio, 'mV, CurrIO=',iio, 'mA'
print(str(epoch)+":"+str(charge)+":"+str(temp)+":"+str(vbat)+":"+str(ibat)+":"+str(vio)+":"+str(iio))
</pre>

<pre>
#!/usr/bin/python
# pijuice-set_charging.py

import calendar
import time
import sys
from pijuice import PiJuice

pijuice = PiJuice(1,0x14)

status = pijuice.status.GetStatus()
status = status['data'] if status['error'] == 'NO_ERROR' else status['error']

charging_enabled = pijuice.config.GetChargingConfig().get('data', {}).get('charging_enabled')

if len(sys.argv) != 2:
        print charging_enabled
else:
        if int(sys.argv[1]) == 1:
                pijuice.config.SetChargingConfig({'charging_enabled': True}, True)
        else:
                pijuice.config.SetChargingConfig({'charging_enabled': False}, True)
</pre>

Then, in crontab,
<pre>
*/10 * * * * P="`sudo pijuice-scan.py | cut -d':' -f2`"; [ $((P)) -lt 40 ] && sudo pijuice-set_charging.py 1 && echo ${P} "\% in charging"
*/15 * * * * P="`sudo pijuice-scan.py | cut -d':' -f2`"; [ $((P)) -ge 97 ] && sudo pijuice-set_charging.py 0 && echo ${P} "\% in discharging"
</pre>

Suppose you would have better and smarter solution, perhaps can suggest in README.md

Thanks for listening to.

tvoverbeek, 2018-08-14:
Looks good.
With this approach (Check if charging needed every 10 minutes and check to finish charging every 15 minutes) you do not need the sys task (the pijuice service) active. Only the pijuice module is needed.
What do you use to charge in this scenario? Solar array?

leeee-me, 2018-08-15:
My intention is simple. My RPi3B always has a power adapter plugged in, and then I assume the continuously charging would raise the temperature of battery and system and probably damage the battery gradually. However, I found the crontab mechanism is not working precisely as expected (bouncing value of charging level?), Suppose the any meter/gauge monitor by Firmware or pijuice.service could be better. By the way, I noticed that gpio_in_enabled also influence the charging behavior. Sorry I am not an expert of power/battery system. Here is another implementation:
<pre>
#!/usr/bin/python
# pijuice-set_charging.py

import calendar
import time
import sys
from pijuice import PiJuice

PID_FILE = '/var/run/pijuice.pid'

pijuice = PiJuice(1,0x14)

def notify_service(*args):
    ret = -1
    try:
        pid = int(open(PID_FILE, 'r').read())
        ret = os.system("sudo kill -SIGHUP " + str(pid) + " > /dev/null 2>&1")
    except:
        pass
    return ret

status = pijuice.status.GetStatus()
status = status['data'] if status['error'] == 'NO_ERROR' else status['error']

if len(sys.argv) > 1:
        if int(sys.argv[1]) == 1:
                pijuice.config.SetChargingConfig({'charging_enabled': True}, True)
                pijuice.config.SetPowerInputsConfig({'gpio_in_enabled': True}, True)
        else:
                pijuice.config.SetChargingConfig({'charging_enabled': False}, True)
                pijuice.config.SetPowerInputsConfig({'gpio_in_enabled': False}, True)
        notify_service()

charge = pijuice.status.GetChargeLevel().get('data', {})
charging_enabled = pijuice.config.GetChargingConfig().get('data', {}).get('charging_enabled')
gpio_in_enabled = pijuice.config.GetPowerInputsConfig().get('data', {}).get('gpio_in_enabled')
print "charging_enabled: "+str(charging_enabled)+", gpio_in_enabled: "+str(gpio_in_enabled)+", charge: "+str(charge)+"%"
</pre>

Strangely, the gpio_in_enabled cannot be altered/changed by pijuice py module. Did I make any mistake?
<pre>
~$ sudo pijuice-set_charging.py 1
charging_enabled: True, gpio_in_enabled: True, charge: 97%
~$ sudo pijuice-set_charging.py 0
charging_enabled: False, gpio_in_enabled: True, charge: 97%
~$ sudo pijuice-set_charging.py 1
charging_enabled: True, gpio_in_enabled: True, charge: 97%
</per>

leeee-me, 2018-08-15:
Or I tried this logic but it seems to have no effort/no use.
<pre>
if len(sys.argv) > 1:
        INPUTS_PRECEDENCE = pijuice.config.powerInputs
        USB_CURRENT_LIMITS = pijuice.config.usbMicroCurrentLimits
        USB_MICRO_IN_DPMS = pijuice.config.usbMicroDPMs

        config = {}
        result = pijuice.config.GetPowerInputsConfig()
        if result['error'] == 'NO_ERROR':
                pow_config = result['data']
                config['precedence'] = INPUTS_PRECEDENCE.index(pow_config['precedence'])
                config['gpio_in_enabled'] = pow_config['gpio_in_enabled']
                config['usb_micro_current_limit'] = USB_CURRENT_LIMITS.index(pow_config['usb_micro_current_limit'])
                config['usb_micro_dpm'] = USB_MICRO_IN_DPMS.index(pow_config['usb_micro_dpm'])
                config['no_battery_turn_on'] = pow_config['no_battery_turn_on']

        if int(sys.argv[1]) == 1:
                pijuice.config.SetChargingConfig({'charging_enabled': True}, True)
        #       pijuice.config.SetPowerInputsConfig({'gpio_in_enabled': True}, True)
                config['gpio_in_enabled'] = True
                pijuice.config.SetPowerInputsConfig(config, True)
        else:
                pijuice.config.SetChargingConfig({'charging_enabled': False}, True)
        #       pijuice.config.SetPowerInputsConfig({'gpio_in_enabled': False}, True)
                config['gpio_in_enabled'] = False
                pijuice.config.SetPowerInputsConfig(config, True)
        notify_service()

</pre>

leeee-me, 2018-08-15:
After checked the pijuice_cli.py, the PowerInputConfigs needs to be set in a one complete shot:
<pre>
if len(sys.argv) > 1:
        config = {}
        result = pijuice.config.GetPowerInputsConfig()
        if result['error'] == 'NO_ERROR':
                pow_config = result['data']
                config['precedence'] = pow_config['precedence']
                config['gpio_in_enabled'] = pow_config['gpio_in_enabled']
                config['usb_micro_current_limit'] = pow_config['usb_micro_current_limit']
                config['usb_micro_dpm'] = pow_config['usb_micro_dpm']
                config['no_battery_turn_on'] = pow_config['no_battery_turn_on']

        if int(sys.argv[1]) == 1:
                pijuice.config.SetChargingConfig({'charging_enabled': True}, True)
                config['gpio_in_enabled'] = True
                pijuice.config.SetPowerInputsConfig(config, True)
        else:
                pijuice.config.SetChargingConfig({'charging_enabled': False}, True)
                config['gpio_in_enabled'] = False
                pijuice.config.SetPowerInputsConfig(config, True)

        notify_service()
</pre>
However, when a USB power adapter plugged in PiJuice, It is really hard to check the resultant effect of flags of charged_enable and gpio_in_enabled... Really strange

leeee-me, 2018-08-15:
Issue #190 was created to understand how to implement a true execution plan

closed 2018-08-15 by leeee-me