PiSupply/PiJuice #189: Usage with HiFiBerry AMP2

Leobaillard, 2018-08-14:
Hi!

I'm considering using my PiJuice along with an RPI3B+, and a 7" touchscreen to create a car head unit with Android Auto (Open Auto). I currently have almost everything as I want as a prototype except the sound. The idea is to get rid of the car's stereo. This is why I was thinking of using HiFiBerry's AMP2 and connecting the output to my car's speakers.

My question is related to power supply: I see that the HiFiBerry comes with it's own power jack and is supposed to power the Pi and accessories, but will it be able to power the Pi, the touchscreen and the PiJuice? Moreover, will this way of supplying power interfere in any way with the PiJuice's features? (What I'm most interested in is wake up on charge and power off when there is no power as I'm using the car's 12V line)

I saw the article on Rick Tonoli's blog but he's using an external amplifier, so not completely the same.

Thanks for any information you might be able to provide for this project!

Leobaillard, 2018-09-01:
If anyone is also interested in implementing this solution, I've tried and it works flawlessly. I'm not sure if this setup has any particular adverse effects on the PiJuice or it's battery, but it at least works as expected.

One side note for anyone wanting to try this: you might want to add a diode to prevent 5V to be outputed to the car's electrical system while the Pi is shuting down and the PiJuice has not yet shut the regulator off.

tvoverbeek, 2018-09-02:
@Leobaillard Where can we find a description of your setup?

Leobaillard, 2018-09-02:
Unfortunately nowhere at the moment, I don't have a blog or anything like that. But if writing something up or building schematics could interest someone, I can whip up something. Do you have a place in mind where that would make sense?

tvoverbeek, 2018-09-02:
Just a short description of how you power your setup.
I am assuming you have stacked the RPI, PiJuice and the HiFiBerry AMP2 and the 12V from the car are fed to the barrel connector of the HiFiBerry AMP2.
Since both the HiFiBerry and the PiJuice use I2C I guess there are no conflicting addresses there.
Does the HiFiBerry behave as an HAT (i.e. does it have an id-eeprom on address 0x50)?
If so, the PiJuice id-eeprom address should be changed to 0x52 to avoid conflicts.
When the 12V goes off the PiJuice supplies the RPi for a while. Correct?
I do not understand how the PiJuice 5V can be fed back into the car. Via the barrel connector?
Just correct this description so we have a record on Github. Main interest is in how PiJuice and other boards can work together and put this as an example in the documentation.

Leobaillard, 2018-09-02:
I've stacked them in that order, yes. I use the screw terminals to feed in the +12V, but that should be the same as the barrel connector.

I'm using SYS_FUNC_HALT_POW_OFF on the "No power" system event so that when the car's ignition is turned off, the PiJuice sends the halt command to the Pi and shut down the regulator after 60 seconds. I also use the "Wake up on charge" system task at 0% to turn on the Pi as soon as 12V power is applied.
When I first connected the setup to the car for testing, I noticed that the +5V from the PiJuice was fed back to the car because I also have an outside temperature sensor connected to an Arduino and screen, powered off the car's +12V as well (using a DC-DC converter), and it was still getting power after the ignition was turned off. Moreover, when I turned the ignition off, the low battery light was lit on my dashboard (and I mean with the key completely out, meaning there should not be any light).

My car is quite old (1999) so newer cars might have protection circuits against current flowing into the car from accesories, I don't know. But fixing this was as simple as adding a diode on the power input to the AMP2.

As for the id-eeprom address, I did not know about that. I haven't noticed any problem at the moment, but I'll be sure to change it according to your suggestion. I'm not sure if the AMP2 has one, but I guess it would make sense. Their website states:
> Pin 27 and 28 are always reserved for an ID EEPROM on the Raspberry Pi. Independently which card you use, these pins are always reserved and should never be used to connect external components.
But to me, this doesn't clearly say that the AMP2 is indeed using these pins. The documentation is a lot less abundant compared to the PiJuice!

Apart from all that, the setup seems to function really well for now. I've added a fan controlled by a relay set up to turn on when the PiJuice's battery or the CPU gets too hot. It's not much, but I hope it'll protect the battery a little bit.

Just for reference, in terms of power consumption, here is what is connected to the Pi/PiJuice:
- Raspberry Pi 3B+
  - USB : my smartphone to use with Open Auto
  - USB : an EasyCap composite capture device for the reversing camera
- PiJuice with standard battery
- HiFiBerry AMP2
- A 5V relay
- An IR receiver
- The official Raspberry 7" touchscreen display (connected via the GPIO header)

All of this consumed between 0.3A to 0.7A at 12V on my bench power supply. During my tests, the status panel of the PiJuice, when 12V power was present, always indicated good readings in terms of power stability, consumption and temperature.

In the interest of information gathering, tell me if you'd like me to run some specific tests.

Thanks again for your input!