PiSupply/PiJuice #191: Can't find RTC on PiJuice

richardquint, 2018-08-15:
Following up on issue #185, I found that I can't access the RTC on one of the PiJuice hats.  On one hat I have

pi@Mrs-Beaton:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@Mrs-Beaton:~ $ sudo hwclock -r
pi@Mrs-Beaton:~ $ 2018-08-15 14:34:09.361303-0700
lsmod | grep "rtc"
rtc_ds1307             24576  1
hwmon                  16384  1 rtc_ds1307
pi@Mrs-Beaton:~ $ 

and on the other

pi@Mrs-Lovett:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                  
pi@Mrs-Lovett:~ $ sudo hwclock -r
hwclock: Cannot access the Hardware Clock via any known method.
hwclock: Use the --debug option to see the details of our search for an access method.
pi@Mrs-Lovett:~ $ lsmod | grep "rtc"
pi@Mrs-Lovett:~ $ 

Searching the internet I found https://www.raspberrypi.org/documentation/configuration/device-tree.md#part3 which suggests to me that the PiJuice should provide the rtc_ds1307 module, but it hasn't in the second case. I've tried the SW1 + SW2 reset routine again, but that didn't help.  Is there anything else I can try or could this be a malfunctioning hat?  This hat is also the one which drains the battery in only 45 minutes.

tvoverbeek, 2018-08-16:
Probably in the second case the PiJuice has an ID-EEPROM address of 0x52, which means the OS does not read the HAT ID-EEPROM, since it expects a HAT ID-EEPROM at 0x50.
In the first case the 'UU' in the i2cdetect output means a kernel module is loaded for the RTC, due to the device tree fragment in the pijuice ID_EEPROM..
For the second case either change the ID-EEPROM address back to 0x50 or add a line to config.txt to  explicitly load the i2c-rtc module: `dtoverlay=i2c-rtc,ds1339` and reboot

richardquint, 2018-08-16:
I checked the ID-EEPROM address in the GUI and it was set to be 0x50. Without making any other changes I tried explicitly loading the i2c-rtc module `dtoverlay=i2c-rtc,ds1339` in config.txt and now I have the RTC.  Still unclear why this happened.

tvoverbeek, 2018-08-17:
@richardquint Have you also checked the I2C address of the RTC in the GUI?
The device tree fragment in the ID-EEPROM expects it to be 0x68.
Note the numbers in the GUI are already in hex.
In the GUI the ID-EEPROM address shows as '50' (not 80 decimal) and the RTC address as '68'  (not 104 decimal).

With ID-EEPROM address at 0x50, does `/proc/device-tree/hat` exist?

If not, the HAT ID-EEPROM is not recognized. The ID-EEPROM can be written, so maybe the ID-EEPROM is damaged in that case

richardquint, 2018-08-17:
With the ID-EEPROM address at 0x50,  /proc/device-tree/hat exists.

I have been doing some more physical experimentation and I suspect the problem lies with my combining the PiJuice and the Adafruit PiTFT.  After switching PiJuice hats between the Pis (one with the PiTFT and the other without) I still have the problems with the Pi where I am using both a PiJuice and a PiTFT.  Next I will pull the PiTFT and see if leaves both PiJuices working as they should.  Then we'll know.

tvoverbeek, 2018-08-18:
@richardquint If it is an Adafruit PiTFT with touch then this is most likely the culprit.
The PiTFTs with touch  (not the PiTFTs Plus) use GPIO2 for backlight control.
GPIO2 is also used for the I2C data line (SDA1).
Hence a conflict.

richardquint, 2018-08-18:
@tvoverbeek you beat me to the solution. Last night I pulled the PiTFT off and stopped fbcp which the Adafruit install script put in rc.local and things seem to be running correctly.  Just removing the PiTFT solved the problems with the RTC, but if fbcp is left running it lowers the battery life quite a bit. I'll close this issue.  Thanks for your help.

tvoverbeek, 2018-08-19:
Looking again at the Adafruit docs and their easy-install script it seems the PiTFT plus with resistive touchscreen still claim GPIO2 for the backlight in the dtoverlay.
The PiTFT plus with capacitive touch screen uses I2C to talk to the touch driver chip on address 0x38 according to the dtoverlay.
Which (now broken) PiTFT variant do you have?

shawaj, 2018-08-19:
Product link for reference - https://www.adafruit.com/product/2441

tvoverbeek, 2018-08-20:
@richardquint OK, I think I found what the problem is when you combine the Pijuice with the Adafruit 3.5" PiTFT resistive touch screen (ttps://www.adafruit.com/product/2441).
The PiTFT also behaves as a HAT since it has a 24c32 EEPROM at address 0x50 on I2C bus 0.
Hence the PiJuice EEPROM at 0x50 is not recognized.
Do the following:
- Change PiJuice ID-EEPROM address to 0x52, so there is no conflict with the PiTFT.
- To address the PiJuice RTC from the OS include the line `dtoverlay=i2c-rtc,ds1339` in /boot/config.txt (as you already did). Note this does not make the alarm function to the OS.

richardquint, 2018-08-20:
Thank you so much for your research and help @tvoverbeek. I'll post on how things work out when I get my new PiTFT Plus.