PiSupply/PiJuice #192: PiJuice alternate power supply

kf4jbz, 2018-08-15:
Using PiJuice with supplied Motorola BP7X battery  Off grid use with HAM radio.  BP7X is powering RPi 2 and 7 inch touch screen.
Not enough battery life to run RPi and touchscreen for more than a few minutes.  Found alternate 5v power supply PiPower module at tindie.geppettlelectronics.com.  Issue- Is it safe to attach the Pi Power after removing the PiJuice battery to the GPIO header on top of the PiJuice without destroying the PiJuice hat?
Also removing the PiJuice hat over and over has led to communication errors  with RPi 2.

shawaj, 2018-08-15:
Do you have a link to the product you are talking about?

tvoverbeek, 2018-08-16:
@shawaj  See https://www.tindie.com/products/nsayer/pi-power/ and http://www.geppettoelectronics.com/2014/07/pi-power-user-guide.html for documentation.
including schematic.

tvoverbeek, 2018-08-16:
@kf4jbz Should work, provided there is no mechanical interferece with the PiJuice.
It is equivalent to having USB power connected to the RPi. The PiPower feeds the 5V to the GPIO pins.
Although with the PiPower connected I do not see the need to use the PiJuice.

ChristopherRush, 2019-09-15:
We now stock much larger batteries, which you can find here - https://uk.pi-supply.com/search?type=product&q=pijuice+battery

closed 2019-09-15 by ChristopherRush