PiSupply/PiJuice #193: Not enough clear to illustrate workflow of wakeup_enable.py

leeee-me, 2018-08-16:
The sample file is not clear enough to demonstrate the precise workflow I believe. Each time when system reboot, the wake-up alarm will be cleaned and must be set again via PiJuice py module once if we wish it to be alarmed again. **Besides**, another issue is pijuice_sys.py seems not know this. This can be identified by running pijuice_cli.py and read the wake-up alarm page.

Here is my suggestion of wakeup-enable.py
<pre>
#!/usr/bin/env python
#
# This script is started at reboot by cron
# Since the start is very early in the boot sequence we wait for the i2c-1 device

import pijuice, time, datetime, os

while not os.path.exists('/dev/i2c-1'):
    time.sleep(0.1)

pj = pijuice.PiJuice(1, 0x14)

PID_FILE = '/var/run/pijuice.pid'

def notify_service(*args):
        ret = -1
        try:
                pid = int(open(PID_FILE, 'r').read())
                ret = os.system("sudo kill -SIGHUP " + str(pid) + " > /dev/null 2>&1")
        except:
                pass
        return ret

#sync time
t = datetime.datetime.utcnow()
tt = str(t.year)+"-"+str(t.month)+"-"+str(t.day)+"T"+str(t.hour)+":"+str(t.minute)+":"+str(t.second)+"."+str(t.microsecond)

pj.rtcAlarm.SetTime({'second':t.second, 'minute':t.minute, 'hour':t.hour, 'weekday':(t.weekday()+1) % 7 + 1, 'day':t.day,  'month':t.month, 'year':t.year, 'subsecond':t.microsecond//1000000})

#print(pj.rtcAlarm.GetTime().get('data'))

#set alarm
pj.rtcAlarm.ClearAlarmFlag()
a = {}
a['second'] = 0
a['minute'] = 0
a['hour'] = 7
a['day'] = 'EVERY_DAY'
a['weekday'] = 'EVERY_DAY'
pj.rtcAlarm.SetWakeupEnabled(True)
status = pj.rtcAlarm.SetAlarm(a)

if status['error'] != 'NO_ERROR':
        print "pijuice-wakeup[",tt,"UTC ] SetAlarm Err:", status['error']
else:
        print "pijuice-wakeup[",tt,"UTC ] SetAlarm:", pj.rtcAlarm.GetAlarm().get('data')
        if notify_service() != 0:
                print "pijuice-wakeup[",tt,"UTC ] notify_service failed, try once after 60s"
                time.sleep(60)
                t = datetime.datetime.utcnow()
                tt = str(t.year)+"-"+str(t.month)+"-"+str(t.day)+"T"+str(t.hour)+":"+str(t.minute)+":"+str(t.second)+"."+str(t.microsecond)
                print "pijuice-wakeup[",tt,"UTC ] notify_service again ret:", notify_service()
</pre>

tvoverbeek, 2018-08-16:
@leeee-me The PiJuice service (pijuice_sys.py) is not even aware of the wakeup-enable flag, so I do not see why you need to notify it.
There may be a problem with pijuice-cli not updating the status of wakeup-enable correctly.
Will check that later.

leeee-me, 2018-08-16:
In that case, this concern can be dissolved

closed 2018-08-16 by leeee-me