PiSupply/PiJuice #196: Problems with HARD_FUNC_POWER_ON

bobmorane1, 2018-08-22:
Hi,

i have a problem with my PiJuice.
I configured SW1_LONG_PRESS_1 to a user function like this:

> from pijuice import PiJuice
> import os
> pijuice = pijuice.PiJuice(1, 0x14)
> 
> pijuice.power.SetSystemPowerSwitch(0)
> pijuice.power.SetPowerOff(15)
> os.system("sudo halt")

That works fine so far. The RPi shuts down and after 15 seconds the PiJuice Removes the 5V Power to RPi and will then enter a low power state indicated by a very slow blinking weak led.

To turn everything back on again, SW1_SINGLE_PRESS is configured to HARD_FUNC_POWER_ON.
The Problem is, when I press SW1, it sometimes works fine and the PiJuice LED gets to a solid green, turns on the power to the RPi and the system starts booting and everything is fine.

But some other times when I press SW1, the PiJuice LED gets to a solid green only for a second or two  and then falls back to the very slow blinking weak led and the RPi gets no power. In this case, i have found two options to get power to the RPi again.

1. Use another Button configured to HARD_FUNC_RESET on a single press. -> The RPi gets power and starts booting right away.

2. Use SW1_LONG_PRESS_2 configured to HARD_FUNC_POWER_OFF -> while holding SW1 for about 10 seconds, the PiJuice LED stays solid green (RPi still gets no power) -> release SW1 (PiJuice LED falls back to the slow blinking state) -> Single Press SW1 again and the RPi gets power and starts booting.

Do you have any ideas why HARD_FUNC_POWER_ON sometimes doesn't power up the RPi?

tvoverbeek, 2018-08-22:
Question: What does the red RPi power led do in these cases?

bobmorane1, 2018-08-22:
RPis red power led stays off. It looks like the PiJuice doesn't deliver the 5V to the GPIO Header until i do a HARD_FUNC_RESET or HARD_FUNC_POWER_OFF

tvoverbeek, 2018-08-22:
Other question: Do you have the pogo-pin (run pin) installed and/or defined the RUN PIN as installed on the General tab? 
HARD_FUNC_RESET works differently depending on the run pin installed or not.

bobmorane1, 2018-08-23:
No i have not installed the pogo-pin and the configuration is set to NOT_INSTALLED also.

I think thats the right setting for me, because im powering the system with a 5V 3A DC wall adapter connected to J4 and i want the pijuice to completely cut power to the RPi in Shutdown state.

ChristopherRush, 2019-09-15:
If you still need help with this issue please reopen.

closed 2019-09-15 by ChristopherRush