PiSupply/PiJuice #199: Pijuice_cli broken on Import urwid but then rework after rebooting

ilulissat, 2018-08-25:
Hello There,
I get recently a new pijuice, work perfectly until updating (throught pijuice_cli.py) firmware to 1.2, then now_ i get trouble with something completely different : cli UI. 

root@myraspberrypi:~# pijuice_cli.py 
```
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 11, in <module>
    import urwid
  File "/usr/lib/python2.7/dist-packages/urwid/__init__.py", line 36, in <module>
    from urwid.wimp import (SelectableIcon, CheckBoxError, CheckBox, RadioButton,
ValueError: bad marshal data (unknown type code)

```
EDIT : after rebooting pi, issue is solved. But as far i don't know why the issue happened, i do think this issue is no solved yet.

tvoverbeek, 2018-08-25:
@ilulissat I do not know why this happened.
You did not run a mix of python3 and python2?
See https://stackoverflow.com/questions/30861493/how-to-fix-python-valueerrorbad-marshal-data for some explanations of this error.

ilulissat, 2018-08-25:
I'm just using pijuice from debian package of raspbian. so, i have both python 3.5 and python 2.7.
I just have the issue now after some time, so it is clearly not fixed.

`find /usr -name '*.pyc' -delete`
is working but as say in the tex, that clearly not a good option due to read/write issue.

To avoid all this pydep issues, was it possible to run  pijuice_cli.py from a virtualenv or not at all ?

tvoverbeek, 2018-08-25:
@ilulissat I guess you somehow corrupted the compiled python files (*.pyc) in /usr/lib/python2.7/dist-packages/urwid. This could have something to do with running as root, so you can access/modify any file on the system.
Running things as root is a bad idea in general.

closed 2019-09-15 by ChristopherRush