PiSupply/PiJuice #200: No dir  /home/pi/PiJuice

Pannekoek22, 2018-08-27:
Dears, I have fresh Raspbian + pijuice-gui and want to use the Wakeup alarm for repeatable booting. Then I should use wakeup_enable.py script as mentioned in 'PiJuice Software README.md' but I do not find directory '/home/pi/PiJuice' on my Pi, so also no script: '/home/pi/PiJuice/Software/Test/wakeup_enable.py'. My question is how to get those data on my Pi: do I need to create the dir-tree and do I have to copy the script? I should expect that installing pijuice-gui should do this for me.

Pannekoek22, 2018-08-27:
Yes, this did fix the issue, thanks.

tvoverbeek, 2018-08-27:
@Pannekoek22 Ok to close this issue?

Pannekoek22, 2018-08-27:
Ja, you can close the issue.

Germ.

closed 2018-08-27 by tvoverbeek