PiSupply/PiJuice #201: PiJuice RTC not keeping time

StudioBeneden, 2018-08-31:
Hello all, 

I'm setting and reading time from the PiJuice as per the setup mentioned here (
https://thepihut.com/blogs/raspberry-pi-tutorials/17209332-adding-a-real-time-clock-to-your-raspberry-pi) I'm setting the clock with the 'date' command, since the Pi is running as an AP, without internet connection.

On first reboot after setting the time, the clock is already a few seconds off. After a short while, with some shutdowns and reboots in-between,  the RTC is running a few minutes behind time.
The PiJuice is running from mains, battery around 98%. 

Is this a faulty clock on the PiJuice? Or can someone think of another reason for this drift?

Thanks

StudioBeneden, 2018-08-31:
Hi Ton, thank you for you fast response!

Thing is; I'm doing it exactly as you described, And I've used the same way with several separate RTC modules before without these issues. When checking the i2c address (with sudo i2cdetect -y 1) my PiJuice comes up on 0x68. with the two UU marks, telling it is configured.
I haven't changed the eeprom address. thats still on 0x50.

So:
- I set the date manually, with the 'date' command
- use 'sudo hwclock -w' to write it to the PiJuice hwclock
- in rc.local use 'sudo hwclock-r' to write the time to system again at boot.
when checking the time with 'sudo date; hwclock-r' both times ar the same, but off  

Just to test, I've #-out the  'sudo hwclock-r' in rc.local, assuming the time from both clocks must be apart by ca. 10 seconds after reboot. But even then they are the same...

Meanwhile in the Netherlands:
While testing, I discovered that the battery wasn't charging (no blinking blue light). When I powered down, took it off and rebooted, the PiJuice config still said battery power was on 92%, not discovering that the battery was absent... When I replaced the battery and started up again, the battery was running REALLY hot... So for now, I've switched everything off. I'll contact my reseller to check. Thank you for thinking with me again, I'll keep you posted..

tvoverbeek, 2018-08-31:
@StudioBeneden I have seen some reports that there are PiJuices delivered with no firmware loaded.
This might be the case for you (would explain the symptoms you see).
Does i2cdetect show any response on 0x14?

StudioBeneden, 2018-08-31:
The PiJuice responds on 0x14, (i've connected it without battery now) It used to work quite ok, I've been able to update the firmware to v 1.2 so firmware is not the issue here I guess.. 

**off topic:**
Before I got to struggling with the clock, I was at a point where the PiJuice did everything I wanted it to do: Keeping the pi alive while I was changing it's location, temporarily disconnected from mains. Power down at 10% worked fine (it would keep the Pi alive for around 2,5h)  Restarting when the mains power came on again was a bit flimsy; now and then it would start up 3 or more times before it stayed on. The battery would never reach 'full' but would keep charging and hovering around 98%.  But now if connect the PiJuice to 5V (over USB), even without the battery, the charge IC (or the little buck-convertor next to it..) runs very (abnormally) hot. Something bad happened hardware-wise, I guess..

tvoverbeek, 2018-09-01:
@StudioBeneden Probably time to contact the supplier and/or Pi-Supply directly for a possible replacement. In case you contact Pi-Supply refer to this issue.

StudioBeneden, 2018-09-11:
Supplier (SOS-solutions) exchanged the board without delay. Been testing over the last few days, this one doesn't run hot and stops charging (blue light) when the battery is full (the last one would blink on for hours, even without the Pi connected) I think I had a faulty board/battery earlier.

As for the clock-issues... I'm guessing I didn't properly disable the fake-hwclock on that occasion.. So I might have been looking at the time from the fake-hwclock instead of the real hw one on the pijuice all along. After the new Pijuice arrived I did a fresh install and all is working fine now!

thanks for your help again!

tvoverbeek, 2018-09-11:
@StudioBeneden Closing this issue, seems to work now.
If not happy, please reopen.

closed 2019-09-15 by ChristopherRush