PiSupply/PiJuice #202: PiJuice: no charge, battery voltage fluctuates, sometime 'not connected'

sejtam, 2018-09-02:
I just dug up my PiJuice that I got from the initial kickstarter but never got around to using, and attached it to a Pi3. But it doesn't work. The battery shows fluctuating voltage and no charge
At times I get 'not connected' shown for the battery.

I did switch on 'çharging enabled' (which was turned off initially) but no luck. I tried
adding 5v to both the PI an charge via GPIO5 and to the PiJuice itself, but no charging happens.

I managed to upgrade the firmware to 1.2 but that too did not help.

Could someone help with steps to check what could be the cause of this?

attached some screen shots

[image]
[image]
[image]

shawaj, 2018-09-02:
Have you got i2c turned on?

tvoverbeek, 2018-09-02:
@shawaj I2C should be on, otherwise there would be 'COMMUNICATION ERROR' in the screenshots.
Looks to me like a completely discharged battery.
There is 0.5-0.6A flowing from the RPi to the PiJuice, so it seems there is a charge current.
However the battery voltage is extremely low, far below the cutoff voltage of 3V.

Do not know how the firmware/charging circuit handles charging starting from an extremely low battery voltage.

@sejtam Does anything on the PiJuice get extremely hot when charging?
Negative values on GPIO power input mean current is flowing from the Pi GPIO 5V pin to the PiJuice. Positive values mean the PiJuice is supplying current to the RPi via the GPIO pin.

sejtam, 2018-09-03:
Ton,

The Rpi gets hot, but not extremely I think.

A deeply discharged battery would make sense. How do I get it out of
this state?
I left it connected overnight and it did not change

M

tvoverbeek, 2018-09-03:
@sejtam Since you did not use your PiJuice before, did you remove the plastic tab between the battery and the battery connector?
I get similar behavior as in your screen shots when I remove the PiJuice battery and power via the RPi USB.

sejtam, 2018-09-03:
Thanks, doh. mea culpa... peisei

I'll report how it goes :-)

tvoverbeek, 2018-09-03:
Since your PiJuice seems to be working (See issue #203). Closing this one.

closed 2018-09-28 by tvoverbeek