PiSupply/PiJuice #203: Event for 'Power ids back'?, logging of user_script calls?

sejtam, 2018-09-03:
I found the 'No power' event. How fast does that trigger after the power is gone?

Is there s similar event for when the power comes back (while the Pi is still running)?

How can debug the events and script execution? wIll those get logged somewhere?

ChristopherRush, 2018-09-03:
@sejtam As soon as the primary power is removed the status will change straight away and trigger the "No Power" event. 

For when the power is back you can use the "wake up on charge" under "System Task" tab and set a low value of say 1%. When the PiJuice will detect the battery is charging it will trigger this event.

The PiJuice does not have any event log file. I'm not sure if `journalctl` will output anything, I will check later this week.

sejtam, 2018-09-03:
On 3/9/18 23:38 PM, Christopher Rush wrote:
>
> @sejtam <https://github.com/sejtam> As soon as the primary power is
> removed the status will change straight away and trigger the "No
> Power" event.
>

Hmm. I wote a small email script, stored in in /u0/pijuice/lostpower.py,
chmod(1) it it and it works when called from the shell
Then I assigned it to user_script 1 and set that for the 'No power'
event, enabled events but nothing happened when I turned
off the power on the socket. 
>
> For when the power is back you can use the "wake up on charge" under
> "System Task" tab and set a low value of say 1%. When the PiJuice will
> detect the battery is charging it will trigger this event.
>

But does that trigger even when power come back while the PI had not
even shut down yet?

Say it loses power and goes to battery, and 30 seconds later power comes
back without shut down.
Will there be an event to send an 'all clear'?

> The PiJuice does not have any event log file. I'm not sure if
> |journalctl| will output anything, I will check later this week.
>

ChristopherRush, 2018-09-03:
@sejtam There are some test scripts [here](https://github.com/PiSupply/PiJuice/tree/master/Software/Test) in particular `user_func1.py` which should log enabled events. Try running that and then remove power to check it is triggering the event. Also make sure you put the full path to your script i.e. /home/pi/user_func1.py

tvoverbeek, 2018-09-03:
@sejtam Except assigning your script to USER_FUNC1 you also have to enable the system task (on the System Task tab in the GUI). This enables the part of the pijuice.service which reacts on events sent by the PiJuice. The system task polls the Pijuice once per second for new events, so there can be a delay of about 1 second before the Pi can react on the No power event.

Regarding debugging and logging use the Python logging functions as in user_func1.py as suggested by @ChristopherRush 

You cannot use 'wakeup on charge' to detect the external power is back when the Pi is still running.
The PiJuice will try to restart the Pi on this event when it is off, but does nothing when the Pi is still on.
You will have to do your own polling to detect the return of external power using the GetStatus() function. See https://github.com/PiSupply/PiJuice/tree/master/Software#pijuicestatus-functions.
You could start this polling in your user function detecting the No power event.
Note the polling function must not block the system task so kick it off in the background.

ChristopherRush, 2019-09-15:
Suggestion here - https://github.com/PiSupply/PiJuice/issues/203#issuecomment-418162118

closed 2019-09-15 by ChristopherRush