PiSupply/PiJuice #204: ImportError: No module named 'pijuice'

edmozley, 2018-09-04:
Hi

I copied and pasted the python code to get the status of the PiJuice.  I am a complete novice and after some research found the sample code would not work with Python 3 as the print syntax was bad.  I put some brackets around it which I think is correct:

```
from pijuice import PiJuice # Import pijuice module
pijuice = PiJuice(1, 0x14) # Instantiate PiJuice interface object
print (pijuice.status.GetStatus()) # Read PiJuice staus.
```

However I am getting an error:

ImportError: No module named 'pijuice'

I have installed PiJuice GUI and updated the firmware to 1.2 and re-booted.  My first version of the script was called pijuice.py which was in my documents folder but I have now renamed it to mypijuice.py

[image]

edmozley, 2018-09-04:
Update - I have found that running the script from the command prompt works fine but not from within Python 3 IDLE

[image]

tvoverbeek, 2018-09-05:
@edmozley Where did you install pijuice.py? The package installation only installs it in the python2 directory `/usr/lib/python2.7/dist-packages`.
Compare the output of 'import sys; print(sys.path)' in the command prompt and in IDLE3. They are not identical.

richardquint, 2018-09-05:
I'm a Python newbie.  Does pijuice.py only work in Python2 or could you use it with Python3 if you copy it to the /usr/lib/python3/dist-packages directory?

tvoverbeek, 2018-09-05:
@richardquint  The pijuice.py code works both under Python 2 and 3.
Copying it to /usr/lib/python3/dist-packages directory is enough to get it to work with python3.
Also the other scripts (python_gui.py, python_cli.py, python_sys.py and pijuice_tray.py are compatible with both python2 and python3.

edmozley, 2018-09-06:
Hi,
Thanks very much for the reply.  So if I understand correctly when I type "python" in the command prompt that will run my script using Python 2.  I think as I am a beginner I will just write all my scripts in python 2.

So my next question is how do I launch a Python 2 editor?  When I click on the raspberry at the top left hand corner and then click "Programming" there are several applications but none for Python 2 (that I can see).

Thanks again

tvoverbeek, 2018-09-07:
@edmozley Since you are beginning with Python I would suggest to stick with Python3. In the near future Python3 will be the default on Raspbian and the Raspberry Pi Foundation uses only Python3 in their examples/projects.
To get the PiJuice software working with Python3 do the following in a command prompt:
`sudo cp /usr/lib/python2.7/dist-packages/pijuice.py /usr/lib/python3/dist-packages/pijuice.py`.
There is an other nice editor for Python3 especially for beginners: mu. You find it on the desktop menu Programming->mu. 

If you want to stick with Python2, the IDLE for Python2 is still there. In a command prompt start it with `idle`.

ChristopherRush, 2019-09-15:
We have since moved over to Python3

closed 2019-09-15 by ChristopherRush

Passiday, 2021-11-21:
Perhaps it might be useful for you guys, I installed PiJuice from deb package on Pi OS Lite (Bullseye), and the `pijuice.py` was installed in the following directories:
`/usr/lib/python3/dist-packages` (as a symlink)
`/usr/lib/python3.7/dist-packages` (the real file) the 

However, it did not install under `/usr/lib/python3.9/`, and that was a problem because my project's venv was using python3.9. I had to employ manual fix:
```
$ sudo ln -s /usr/lib/python3/dist-packages/pijuice.py /usr/lib/python3.9/pijuice.py
$ sudo ln -s /usr/lib/python3/dist-packages/smbus.cpython-39-arm-linux-gnueabihf.so 
```

pedrosk, 2021-12-01:
@richardquint  Another solution is to call python3.

As mentioned by @Passiday  and @ChristopherRush the code is in python3 while it is likely that the system default is still python2.7 for example:

```
$ python --version
Python 2.7.16
````

Thus execulte like this:
```
python3 pijuicetest.py
```

Hope this helps.

Ed