PiSupply/PiJuice #205: Find if WakeUpAlarm used in Python

edmozley, 2018-09-06:
Hi
I am beginining to get the hang of using the PiJuice API and also I have learnt how to parse a dictionary object eg:

str(pijuice.status.GetChargeLevel().get('data'))

I am wondering if there is a way of telling if the Pi has been switched on using the WakeUpAlarm or if it has been turned on with the SW1 button.

Thanks

Ed

tvoverbeek, 2018-09-07:
Short answer: No.
However if it is switched on by the WakeUpAlarm, you should know since it has to be programmed before.
Initially no wakeup alarms are set.

edmozley, 2018-09-07:
Ah OK - let me describe the scenario as there may be a better way...

I want the PiJuice to wake up the Pi on a schedule and when it starts up I want it to take a picture, log the battery level and status, email it and then shut down.  I have finished the script to do this.

However if I turn on the Pi by pushing SW1 (so I can use it, modify code etc.) I don't want the script to run.  Or if the script does run I would want to use some if...else logic so all the code gets skipped.

jdotbdot, 2018-09-07:
If your script does **sudo shutdown -h**  or **sudo shutdown -h +2** rather than **sudo shutdown -h now** you get 1 minute (or 2) to type **sudo shutdown -c** to cancel it.
Unfortunately once you issue the shutdown command the Pi will refuse connections (I use ssh, not sure if that's true with a monitor/keyboard) so you would have to have **sleep 60** before the shutdown command

tvoverbeek, 2018-09-07:
@edmozley I am assuming you want to use the repeated wakeup (every xx minutes) for this.
Also since you use email, the Pi is connected to the internet.
With the PiJuice id-eeprom at its default address (0x50) at Raspbian boot the DS1339 RTC driver will be loaded. This driver disables the wakeup alarm at boot, so you will have to enable it at boot.
From the Software README:

>When setting the Wakeup alarm for a repeated wakeup, after the initial reboot the Wakeup enabled capability is disabled due to the Raspbian RTC clock initialisation resetting the bit in the PiJuice firmware. To overcome this you will need to run a script to re-enable the wakeup-enable capability. The script is located in PiJuice > Software > Test > wakeup_enable.py

Also since the Pi is only on for a short time, the system clock will go off since syncing with the internet happens only about every 15 minutes. To get a decent time (in the emails the Pi is sending) copy the RTC time to the system clock at boot. Of course you have to set the RTC to the correct time once.

When you start the Pi with SW1 run a script manually which disables the wakeup:
```
pj = pijuice.PiJuice(1, 0x14)
pj.rtcAlarm.SetWakeupEnabled(False)
```
This assumes your "take a picture etc" script runs long enough to give you time to log in and disable the wakeup alarm.

edmozley, 2018-09-23:
Hi,

Sorry I've been silent these last couple of weeks.  A couple of questions:

1. My RTC clock is one hour out (I am based in London and it has the correct date just one behind). The time at the upper right hand corner of the desktop is correct.  I clicked "Set RTC time" but nothing happens.

2. If I put a massive if.... around the entire script using the minute of the time of day would that work if I set the wake up alarm to every hour.  Something like:

```
import datetime
now = datetime.datetime.now()

if now.minute < = 3:
    # execute script
else:
    # do nothing

```
That way if I turn it on at pretty much any time as long as it's not 1, 2 or 3 minutes past the hour I will be able to use the Pi without any problems.

I chose 3 just in case the Pi was slow starting up for any reason.

tvoverbeek, 2018-09-24:
@edmozley 

> My RTC clock is one hour out (I am based in London and it has the correct date just one behind). The time at the upper right hand corner of the desktop is correct. I clicked "Set RTC time" but nothing happens.

This is exactly as expected. The RTC is kept in UTC. Since you are now on British summertime (which is 1 hour ahead of UTC) you have a difference of 1 hour.

Regarding 2)
Yes. that would work. When you start the RPi by button press after xx:03 it should skip the script and you can do your code modifications, etc.

A correction to my previous comment about time synchronization when connected to the internet.
Assuming you are on raspbian stretch, the timesyncd daemon starts at boot and immediately synchronizes the system time with the internet (using ntp) and copies the UTC equivalent to the RTC.
Then you probably do not need to set the system time from the RTC at boot even for a short activity,
but try it out first.

tvoverbeek, 2018-09-25:
@edmozley The command you are looking for is `hwclock`.
To set the systemclock from the RTC: `sudo hwclock --hctosys`.
`man hwclock` for more details.
Note: this works only if you leave the pijuice id-eeprom address at 0x50 or load the i2c-rtc module from /boot/config.txt.

edmozley, 2018-09-25:
Hello!

My PiJuice is running at 0x52 so it doesn't conflict with another HAT so it looks like I will need option 2 "load the i2c-rtc module from /boot/config.txt".  I am afraid this has prompted more questions!

1. I have located the config.txt file - how do I load the i2c-rtc module to this?

2. By editing the config.txt file do I need to do anything else or does that just give me access to the PiJuice clock and then I have to run sudo hwclock --hctosys when the Pi boots up?

tvoverbeek, 2018-09-25:
@edmozley With the pijuice id-eeprom at 0x52 add the following line to /boot/config.txt:
`dtoverlay=i2c-rtc,ds1339` and reboot. See also /boot/overlays/README.
After reboot you can check with `lsmod` if the rtc-ds1307 module is loaded (it should be).
You can also check the output of `sudo hwclock --debug --show`.
My output looks like this:
```
pi@rpi-stretch-full:~ $ sudo hwclock --debug --show
hwclock from util-linux 2.29.2
Using the /dev interface to the clock.
Assuming hardware clock is kept in UTC time.
Waiting for clock tick...
/dev/rtc does not have interrupt functions. Waiting in loop for time from /dev/rtc to change
...got clock tick
Time read from Hardware Clock: 2018/09/25 13:28:44
Hw clock time : 2018/09/25 13:28:44 = 1537882124 seconds since 1969
Time since last adjustment is 1537882124 seconds
Calculated Hardware Clock drift is 0.000000 seconds
2018-09-25 15:28:43.225526+0200
pi@rpi-stretch-full:~ $ 
```
Note it shows both the hardware clock time in UTC and as last line the time converted to local time.

Then at boot when you start your program (via /etc/rc.local or as cron job or ...) include the `sudo hwclock --hctosys` command to copy the RTC to the system clock.

hp1337, 2018-11-08:
@edmozley I am working with the PiJuice to do exactly what you are. I have been struggling with the same issues, specifically how to debug my code if the PiJuice is turned on with a button rather than with a wake up alarm.

I have settled on the following paradigm:
1. Disable the default pijuice_sys daemon. It becomes too complicated to coordinate the polling of the PiJuice when 2 different scripts are communicating with it at the same time.
2. Write your own daemon that does ONLY what you need.
3. In your custom daemon that runs on boot: Look for a button press of SW3. If SW3 was pressed: Disable reboot. This way you can SSH into your device to debug.
4. I've had situations where something goes wrong and the process breaks down and the Pi will stay running and drain the battery. Thus enable the Raspberry Pi Watch dog to reboot if no heartbeat felt in 15 seconds.

tvoverbeek, 2018-11-08:
@hp1337 You wrote:
> I've had situations where something goes wrong and the process breaks down and the Pi will stay running and drain the battery. Thus enable the Raspberry Pi Watch dog to reboot if no heartbeat felt in 15 seconds.

This covers the case when the RPi goes in an endless loop.
What about using the PiJuice watchdog function? The PiJuice watchdog gets reset whenever you send a command to it. If it times out it will force a reboot (as does the Pi watchdog).

ChristopherRush, 2019-09-15:
I think we have fully documented all aspects of the built-in RTC. If anything is unclear then reopen or raise a support ticket here - https://uk.pi-supply.com/pages/contact-us

closed 2019-09-15 by ChristopherRush