PiSupply/PiJuice #206: No Power Function uses Inputs precedence

ChristopherRush, 2018-09-10:
The "No Power" function under "System Events" uses the "Inputs precedence" setting under "PiJuice HAT Configuration" which by default is set to 5V_GPIO under 1.3.3. For users wanting to use this event whilst connecting power to the PiJuice HAT, they must change the "Inputs precedence" configuration to "USB_MICRO". 

The "No Power" function should detect NOT_PRESENT on both micro USB and GPIO inputs in order to be triggered rather than "Inputs precedence" in my opinion or it just needs documenting to state the above, unless I'm missing something else here?

tvoverbeek, 2018-09-11:
@ChristopherRush Where/How did you find this?
The no power event is generated in the service (pijuice_sys.py) when both the USB and GPIO inputs have status 'NOT PRESENT' or 'BAD'.
See function _EvalPowerInputs() in pijuice_sys.py.
The 'Input precedence' is not used.
Tested it with Input Precedence both ways and it does not make any difference.
The code does already what you propose.

ChristopherRush, 2018-09-11:
@tvoverbeek I came across this issue yesterday when trying to help a customer out on Discord. I did a fresh install and try to use the "No Power" function to shutdown the RPi when the mains power was removed or switched off from the PiJuice micro USB. For some reason It didn't work and I couldn't work out why until I changed the Input precedence setting from 5V_GPIO to USB_MICRO and then it worked straight away. I will try again to replicate what I did and see the outcome.

tvoverbeek, 2018-09-11:
@ChristopherRush Some additional info. Because the 'no power' event is generated by pijuice_sys it can take up to 3 a 4 seconds before the function assigned to the event triggers.

ChristopherRush, 2018-09-11:
@tvoverbeek For some reason after I apply the "No Power" settings it requires a reboot in order for the changes to be made to the PiJuice.

tvoverbeek, 2018-09-13:
@ChristopherRush A fix is now in the master branch for pijuice_sys.py. See commit https://github.com/PiSupply/PiJuice/commit/97c73078acb41680f1696e8ee3bd2ec04010b6ba

ChristopherRush, 2019-09-15:
See above

closed 2019-09-15 by ChristopherRush