PiSupply/PiJuice #207: Script doesnt run when programmed to switch

tampadrones, 2018-09-11:
I am attempting to run a python script that will copy data from my usb drive to my internal drive. I have made the script executable and even made root own the file. When I press my button, sudo service pijuice status shows: 

pi@raspberrypi:/usr/bin $ sudo service pijuice status
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: 
   Active: active (running) since Tue 2018-09-11 15:17:04 EDT; 38min ago
 Main PID: 445 (pijuice_sys.py)
   CGroup: /system.slice/pijuice.service
           └─445 /usr/bin/python /usr/bin/pijuice_sys.py

Sep 11 15:17:10 raspberrypi pijuice_sys.py[445]: sh: 1: RELEASE: not found
Sep 11 15:27:48 raspberrypi pijuice_sys.py[445]: sh: 1: PRESS: not found
Sep 11 15:27:49 raspberrypi pijuice_sys.py[445]: sh: 1: RELEASE: not found
Sep 11 15:28:45 raspberrypi pijuice_sys.py[445]: sh: 1: RELEASE: not found
Sep 11 15:40:11 raspberrypi pijuice_sys.py[445]: sh: 1: RELEASE: not found
Sep 11 15:41:53 raspberrypi pijuice_sys.py[445]: sh: 1: RELEASE: not found
Sep 11 15:42:39 raspberrypi pijuice_sys.py[445]: sh: 1: RELEASE: not found
Sep 11 15:44:05 raspberrypi pijuice_sys.py[445]: sh: 1: RELEASE: not found
Sep 11 15:44:43 raspberrypi pijuice_sys.py[445]: sh: 1: RELEASE: not found
Sep 11 15:45:44 raspberrypi pijuice_sys.py[445]: sh: 1: RELEASE: not found
lines 1-17/17 (END)

tampadrones, 2018-09-11:
sudo service pijuice status

maxdid, 2018-09-11:
This is my output:
`pi@raspberrypi:~ $ sudo service pijuice status
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-09-11 22:01:18 CEST; 1h 0min ago
  Process: 1686 ExecStopPost=/usr/bin/pijuice_sys.py stop (code=exited, status=0/SUCCESS)
 Main PID: 1694 (pijuice_sys.py)
   CGroup: /system.slice/pijuice.service
           └─1694 /usr/bin/python /usr/bin/pijuice_sys.py

set 11 22:01:18 raspberrypi systemd[1]: Started PiJuice status service.
pi@raspberrypi:~ $
`
... And it returns to prompt

maxdid, 2018-09-12:
@tvoverbeek 
Thanks for your time. Here my settings:

**User Func definition**
` "USER_FUNC1": "sh ~/instabot/examples/ig_outsidethebox/ig_outsidethebox.sh",`

The content of the ig_outsidethebox.sh is
`cd ~/instabot/examples/ig_outsidethebox/`
`python ultimate.py`

**Settings for SW3**
[image]
[image]

About the user script, it's a python script that I can run without any problem from my shell running the command
`sh ~/instabot/examples/ig_outsidethebox/ig_outsidethebox.sh`

Thanks again

tampadrones, 2018-09-12:
@tvoverbeek Here is the script I am trying to run. It basically copies files from the usb drive to the pi. https://pastebin.com/6TDbnHar

USER FUNCTION SCRIPT: /autodrive.py

[image]

tampadrones, 2018-09-12:
I think I solved my problem. I guess I didnt look close enough to the physical switch assignment. Looking at the PiJuice Board, the arrangement is SW2 SW3 SW1. So essentially I have been pressing the wrong button. Since nothing was mapped to that SW3, nothing was happening. I updated my test script with the one I want to run and it is running successfully. I hope this helps someone!

tvoverbeek, 2018-09-13:
@maxdid One thing to check is if the system task is enabled. If not, button events are not processed by the pijuice service and hence no user scripts are executed.
Since you seem to be using the pijuice_cli check the contents of /var/lib/pijuice/pijuice_config.JSON.
Here the start of mine:
```
{
  "system_task": {     <----
    "watchdog": {
      "enabled": true,
      "period": "4"
    },
    "enabled": true,   <----
```
if "enabled" is false, change it to "true".

I would start with a simple python script which logs something to a file.
Look for example to https://github.com/PiSupply/PiJuice/blob/master/Software/Test/user_func1.py
which logs date/time and which event triggered it.
That way you can check if the script gets called.

From the way you call your ultimate.py the extra 'sh' layer can be eliminated.
Set the first line of your python script to `#!/usr/bin/env python` and make the script executable:
`chmod +x ~/instabot/examples/ig_outsidethebox/ig_outsidethebox.sh`.
The change working directory can be done inside the python script:
```
import os
os.cwd('/home/pi/instabot/examples/ig_outsidethebox')
```
Note: Using the '~' for the home directory only works in the shell.
Then your USER_FUNC1 definition becomes:
`"USER_FUNC1": "/home/pi/instabot/examples/ig_outsidethebox/ultimate.py"`

One other thing, while your user script executes the pijuice service is blocked (script is executed as part of the (nominally) 1 second polling cycle)

Hope you get it going

maxdid, 2018-09-13:
@tvoverbeek 
The line `"enabled": true,`   was absent from my configuration but it doen't work!!
I also changed the script as you suggested...

I'll try again this evening with a new clean setup and the example from date logger.
Thanks and I'll keep this thread updated

HKHH, 2018-09-21:
I have the same problem. The button with the user function script doesn't work!
Firmware V1.2
Apps V1.33
I2C Adrress1 = 10

tvoverbeek, 2018-09-22:
@HKHH With I2C Address1 = 10 nothing will work, unless you change the software.
All the utilities (pijuice service, pijucie_cli and pijuice_gui) have hardcoded the PiJuice I2C address as 0x14. Change the address to 14 (the gui/cli assume hex addresses) and see if that improves the situation.

shawaj, 2018-09-22:
@tvoverbeek - do you think there is any merit / possibility in having the i2c address not hard coded or having a possibility to override the hard coding easily on that? Just thinking out loud but could be handy.

Guess it currently relies on data from the hat itself?

tvoverbeek, 2018-09-22:
The i2c address is already settable from the GUI/CLI. The value could be a new parameter in the JSON file which has to be read by all programs.
Problem is when you change it in the GUI/CLI, the service, the tray app and the cli/gui all have to reinitialize their pijuice definitions. Maybe to simplify things, require a reboot after a change or setting it manually in the JSON file?

tvoverbeek, 2018-09-22:
@shawaj See also issue #140 where milan also suggests how to handle changing i2c address.

tvoverbeek, 2019-01-08:
@Madko The daemon, pijice_cli and pijuice_gui running as user pijuice has only been around since package version 1.4 which was released recently.
Apologies for the not up-to-date documentation.
To be precise, user scripts have to be executable and the script owner has to be a member of the pijuice group. The default user (pi) is already a member of group pijuice.
User scripts owned by user pi will run as before.
Will try to update the documents within the next few days.
PS I am only a volunteer assisting Pi-Supply

tvoverbeek, 2019-01-08:
Small correction. The scripts will only run if the script owner is a member of the pijuice group.
To add a non-pi user to the pijuice group: `sudo addgroup <username> pijuice`

ChristopherRush, 2019-09-15:
Running scripts are well documented now.

closed 2019-09-15 by ChristopherRush