PiSupply/PiJuice #208: From Mon-Fri wake up and shut off in specified times

DrumSergio, 2018-09-15:
Hi there! Sorry to bother you. I'm a bit new with the PiJuice. I am using it with a 3B+

I've seen in the documentation that the PiJuice has an Alarm function to wake the Pi. Seems working. But for example, what I am trying to achieve is the Pi to be switched on from 9:00 am to 5:00 pm. Using the Weekday function I can get it to switch it on at 9:00 am every Monday to Friday.

But I would like to know what would be the best way to switch it off at 5:00 pm. Maybe a simple cron job? Set it to start at 5 pm to call a python script that does:

```
pj.power.SetPowerOff(20)
subprocess.call(["sudo", "poweroff"])
```

Or is there something inside the PiJuice scripts already that does that? If it is not, it would be a great addition, so from Mon-Sun you can set at which time to start and shutdown you Pi.

tvoverbeek, 2018-09-16:
@DrumSergio Yes a cron job would be the best way.
I would also add the line `pj.rtcAlarm.SetWakeupEnabled(True)` before the `subprocess.call(["sudo", "poweroff"])`.
This to be sure the wake-up has not been disabled by the RTC code in Raspbian.
Something to think about is how to get the correct time after boot.
If you Pi has an internet connection the timesyncd daemon will update the time.
If you are running without internet then you have to copy the RTC time to the system clock at boot.
Without internet your system time stops when off and starts where it left off at the next boot (fakehwclock module).

DrumSergio, 2018-09-27:
Thank you very much!

DrumSergio, 2018-10-10:
Only one more question, @tvoverbeek 

What would be the best way to set "to boot the Pi up 9:00am every weekday" without entering the GUIs? 
I think via the PiJuice JSON config file can't be done, isn't it?
Could it be done via script? Like `pj.rtcAlarm.SetWakeUp("weekday", "9:00")` or something similar?

Thank you!

tvoverbeek, 2018-10-16:
@DrumSergio 
Use the following function from the pijuice module: `pj.rtcAlarm.SetAlarm()`.
This function is also used when you set the alarm from the gui or cli.
When selecting weekday you can also specify a sequence with `;` as separator.
Weekdays are 1=Sunday, 2=Monday, ..., 7=Saturday.
To specify Mo-Fr use `2;3;4;5;6` for weekday.
By the way you can also do this for hour, e.g. `9;15;20` meaning wakeup at 9:00, 15:00 and 20:00.
For your example:
```
import pijuice

pj=pijuice.PiJuice(1,0x14)
pj.rtcAlarm.SetAlarm({'second': 0, 'minute': 0, 'hour': 7, 'weekday': '2;3;4;5;6'})
```
I specified hour as 7 since I am in Europe, currently UTC+2. It corresponds to 9 local time.
Alarm times have to be specified in UTC.

You still have to make sure the wakeup flag is enabled.
If your RPi is connected to the internet I would do the following:
- change the ID EEPROM address to 0x52, so the ds1339 driver does not get loaded.
- in /boot/config.txt add the line `dtoverlay=i2c-rtc,ds1307`
  This loads the ds1307 variant which does not have an alarm function and hence the wakeup
  flag is not reset by the OS. System time is set from the internet at boot by timesyncd and
  the RTC is updated from the system time.
  Both system time and RTC time (in UTC) will then be correct after a long power off.

closed 2019-09-15 by ChristopherRush