PiSupply/PiJuice #209: Test the watchdog

DrumSergio, 2018-09-15:
Hi!

Simple question. Is there any command possible ins Raspbian so that I can test the watchdog is running OK? Some kind of command that blocks the system to not answer the PiJuice module.

Thank you!

tvoverbeek, 2018-09-16:
Not a simple command.
Any I2C transaction to the PiJuice (0x14) or its simulated RTC (0x68) will reset the watchdog.
I2C transactions are done regularly by the pijuice_sys service, the gui/cli, the tray app and by the system to the simulated RTC. 
When the pijuice service is stopped (either by shutdown or by systemctl command) it disables the watchdog.
Killing the pijuice service does not help, because the service is immediately restarted.
The watchdog setting is only transferred to the PiJuice when the service is started.

Here is a way how you can test it:
- Make sure power is connected to the PiJuice USB, not the RPi USB.
- In the gui/cli change the id-eeprom address from 0x50 to 0x52 and reboot. This makes the system ignore the simulated RTC since the device-tree fragment in the id-eeprom is not processed.
- In /usr/bin/pijuice_sys.py comment out lines 219-231 (= the code to disable the watchdog when the service is stopped).
- In the gui/cli enable the watchdog and set it at e.g. 2 minutes.
- Restart the service `sudo systemctl restart pijuice.service` and check with `sudo systemctl status pijuice.service`.
- Kill the pijuice_tray.py app if using the desktop.
- Exit cli/gui if in use.
- Stop the service: `sudo systemctl stop pijuice.service` and check again with `sudo systemctl status pijuice.service`.

Watchdog reboot should occur after 2 minutes. At least it does for me.

closed 2018-09-27 by DrumSergio

dima72, 2020-09-25:
Hello, 
since Sep 2018, I believe  **/usr/bin/pijuice_sys.py** has change a lot (**lines 219-231**)

what lines to fix it will be now?

tvoverbeek, 2020-09-25:
Current source is on github. Science you reference the issue I assume you
know where to find it.