PiSupply/PiJuice #210: Keep-alive load in HARD_FUNC_POWER_OFF

itlobster, 2018-09-16:
Hi,

I'm working on a project with a pi3B+Pijuice connected to a powerbank. And I would like to keep the powerbank alive ( if it goes off I need to manually press the on button on the powerbank for it to work again later wich is not desirable/possible) when the pi+pijuice are in "HARD_FUNC_POWER_OFF".

Off course I could let the pi+pijuice in "SYS_FUNC_HALT" which draws power  but it draws too much for me (~125mAh per hour at ~5V which is 6000mAh to stay alive 48h) .
Below picture is a measure of power consumption (pijuice battery was at 100% when i started shutdown and acquisition) in "SYS_FUNC_HALT" mode:

[image]

I heard other pi HATs have this "keep-alive load" function which draws very little yet keep the battery alive. I couldn't find it on the pijuice, am I missing something ?

Thank you

tvoverbeek, 2018-09-16:
@itlobster Do you know what the minimum current draw is to keep your power bank on?
When in SYS_FUNC_HALT_POW_OFF or HARD_FUNC_POW_OFF the PiJuice MCU is still on but in a low power mode. The needed current is too low for your power bank to stay alive (they are usually tuned to charge phones and switch off if the charge current is too low).
What you could do is to connect some extra load (resistor) to e.g. the Vsys output on P3 or J3 so the current draw by the PiJuice is above the switch off limit of your power bank. Of course this extra load is then also present when the Pi is on.

itlobster, 2018-09-16:
@tvoverbeek thank you for your answer, but the resistor would consume continuously even when the pi is on ? 
I would say the load to keep the battery alive would be around 100mA but the trick with the keep-alive load is not to have it continuously. For example a hypothetical keep-alive load would be to draw 150mA for 0.1s every 3s, effectively consuming  (150*0.1*60/3*60)/3600 = 5maH per hour, which is 240mAh for 48h and not 6000mAh for 48h.

tvoverbeek, 2018-09-16:
@itlobster As far as I can see right now, you would need some external circuit implementing this keep-alive load (small arduino ???) since the RPi cannot do it while it is off.

itlobster, 2018-09-16:
@tvoverbeek  but the pijuice could, even if not in native features maybe with soldering a small piece of electronics done for that purpose ?

shawaj, 2018-09-16:
@itlobster I think we can probably add this to the PiJuice firmware at some point. But not sure on a timeline. Will ask @mmilann

itlobster, 2018-09-16:
@shawaj great, I guess it means the Pijuice is ready for such feature without hardware addition ? Until it's officially implemented how can I implement it on my own ?

tvoverbeek, 2018-09-16:
@itlobster @fovea1959 You could try to implement the pulsing load with something like an Attiny85 (see e.g. adafruits trinket https://www.adafruit.com/product/1500 which is available for both 3.3 and 5V) where one of its digital outputs switches your extra load via a transistor.
You could use the PiJuice 2 digital outputs on the P3 header which also has 5V and 3.3V.
These outputs could be programmed by some user function (e.g switched on or off) at shutdown or boot.
Have not thought this through completely, just to give you some ideas how to start.

shawaj, 2018-09-16:
I don't see why we couldn't have it as a firmware update but I need to check in more detail. Will post back here when more info

tvoverbeek, 2018-09-16:
@shawaj You will need to modify the board to accommodate the extra load resistor. We are talking about pulsing 150-200 mA.

shawaj, 2018-09-16:
Sorry I misunderstood on first read what was being requested. I should have paid more attention.

What could be possible is an add on board to do this that connects into the p3 6 pin header that you mention. Could be pretty useful and relatively inexpensive.

itlobster, 2018-09-17:
@tvoverbeek @shawaj  what would you buy and how would you put it on the board and program it ? If you can draft a little "step-by-step guide" I can buy and test it and come back to you with the results.

Since it might be useful for some people with similar projects to know, below is the measured consumption of power drain in "HARD_FUNC_POWER_OFF" mode over an hour (pijuice battery was at 100% when I started acquisition) which was a steady ~15.9 mAh per hour.
[image]

tvoverbeek, 2018-09-17:
@itlobster You could connect a load resistor between Vsys and GND on P3. The Vsys output can be enabled/disabled by the PiJuice MCU.
To get a pulsing load this would have to be commanded by the PiJuice since the RPi is off.
@shawaj This could be done with a firmware change which switches Vsys on and off with a TBD duty cycle when the 5V to the RPi is off. No hardware change needed to the board.
@itlobster Right now something else has to command the Vsys on and off when the RPi is off.
This could (maybe) done by a small Arduino clone using the needed I2C commands. Do not know if this will work. Have to make sure both I2C masters are not active together. See pijuice.py Set SystemPowerSwitch function for the needed commands.
Will try to test this when I have some time, since I have some small Arduino clones lying around, but right now I cannot promise when this will happen.
@shawaj Can you check with milan how much work this firmware change would be? Probably some cli/gui additions are also needed to enable/disable this feature and set the on and off times.

itlobster, 2018-09-17:
@tvoverbeek seems to be complicated, and it would maybe be less expensive to purchase an usb load extension like  [this one]( https://www.sotabeams.co.uk/usb-battery-pack-keep-alive-load/) (but I'm not sure it's a pulsing one). Anyway it takes space and <del>is a pain not to have it</del> would be nice to have it on the pijuice itself.
The pulsing dummy load clearly is a reason one would buy a witty pi 2 and not a pijuice so I think it is likely to be a high priority firmware change if it's feasible.

tvoverbeek, 2018-09-17:
@itlobster I have no idea how many PiJuice users are powering it from a power bank.
I find it a bit strange to charge the PiJuice battery from another battery, but I understand it if you want to use the PiJuice for things like scheduled wake up and shutting down if power is almost out which the power bank does not do.
I agree the firmware change would be the simplest solution.

mmilann, 2018-09-17:
I agree with approach  @tvoverbeek <https://github.com/tvoverbeek> suggested
with load resistor that will be switched on/off having periodic load
pulses. But Vsys switch has high capacitive load  ~10uF and should not be
switched periodically it can overheat. Instead use N-Channel mosfet and
drive it with IO1 or IO2 from P3 header on PiJuice and then configure it to
generate PWM (from GUI or python init script) with minimum duty circle that
will keep alive pack (PWM period set to maximum ~131ms). So in This case
simple circuit consists of N-mosfet (300mA+, 20V+) with source connected to
P3-GND pin, gate connected to IO1-pin, drain connected to load resistor
(resistance choose to drain enough current from source), other end of
resistor connected to + lead of pack (alternatively can be connected to
Vsys turned on constantly).

tvoverbeek, 2018-09-17:
@mmilan on a side note. Your pijuice-soc-test script switches Vsys continuously (0.5 sec off, 1.5 sec on). Hope I did not damage it when running my discharge test overnight.

itlobster, 2018-09-18:
I am not an expert on the electronic part so I'll just ask the following questions for me and my fellow pijuice users:
1. Is keep-alive load doable with no hardware addition ?
2. If yes, when will it be implemented in the firmware ?
3. If no, what should we buy ?

That's about as down to earth as it gets, hope it's not rude to be so blunt there ;) .

tvoverbeek, 2018-09-18:
Trying to answer your questions:
> Is keep-alive load doable with no hardware addition ?
Short answer No, you need at least an external load resistor and N-Channel mosfet (See @mmilann comment above). In addition a firmware change is needed.
> If yes, when will it be implemented in the firmware ?
The needed firmware change is being considered, but I am only a volunteer so I cannot commit to anything.
> If no, what should we buy ?
Sine you are using a power bank as main source I would follow your own suggestion: the Witty Pi2 or Witty Pi Mini. Both provide a RTC, scheduled wake up, and a dummy pulsing load.