PiSupply/PiJuice #212: Periodic wakeUp and Powerdown not relaible

topshed, 2018-09-18:
I'm trying to have my Pi wake up periodically, do some work and then power down. I'm using a Stretch Lite o/s and the Python3 script below. 

I can get things to work for a few cycles, then it stops. I'm seeing lots of communication errors when reading from the PiJuice so I've added retries when getting or setting.  This has improved the situation a little, but I still haven't been able to get PiJuice to work consistently over more than 4 cycles.

```
from __future__ import print_function

import pijuice
import subprocess
import datetime
import os
import sys
import logzero
from logzero import logger
from subprocess import call
from gpiozero import Button
from time import sleep
import time

DELTA_MIN=5

logzero.logfile("/home/pi/solartest.log")

b = Button(21,pull_up=False)

subprocess.call(["sudo", "hwclock", "--hctosys"])

# Record start time
txt = datetime.datetime.now().strftime('%Y-%m-%d %H:%M:%S') + ' -- Started\n'
logger.info(txt)

while not os.path.exists('/dev/i2c-1'):
    sleep(0.1)

try:
    pj = pijuice.PiJuice(1, 0x14)
except:
    logger.info("Cannot create pijuice object")
    sys.exit()

logger.info("waiting (for network)")
sleep(30)
logger.info("Taking measurements")

# Do stuff

# setting alarm
logger.info('Setting alarm')
a={}
a['year'] = 'EVERY_YEAR'
a['month'] = 'EVERY_MONTH'
a['day'] = 'EVERY_DAY'
a['hour'] = 'EVERY_HOUR'
t = datetime.datetime.utcnow()
a['minute'] = (t.minute + DELTA_MIN) % 60
a['second'] = 0
tries = 0
while tries < 5:
    status = pj.rtcAlarm.SetAlarm(a)
    if status['error'] != 'NO_ERROR':
        logger.info('Cannot set alarm')
        tries+=1
        if tries > 4:
            sys.exit()
        sleep(1)
    else:
        logger.info('Alarm set for ' + str(a))
        tries = 5

# setting wakeup
logger.info('Setting wakeup')
tries = 0
while tries < 5:
    set_status = pj.rtcAlarm.SetWakeupEnabled(True)
    if set_status['error'] != 'NO_ERROR':
        logger.info('Cannot set wakeup' + str(tries))
        tries+=1
        if tries > 4:
            sys.exit()
        sleep(1)
    else:
        logger.info('Wakeup set')
        tries = 5
sleep(0.4)
# measuring charge
logger.info("reading charge")
tries = 0
while tries < 5:
    try:
        charge_stat = pj.status.GetChargeLevel()
        if set_status['error'] == 'NO_ERROR':
            pj_charge = charge_stat['data']
            tries = 5
        else:
            tries+=1
    except KeyError:
        tries+=1
        logger.info("KeyError reading charge" + str(tries))
        sleep(1)
    except:
        logger.info("error reading charge")
        tries = 5

logger.info("Checking button")
if b.value == False:
    logger.info("setting PJ to poweroff")
    pj_off = 'failed'
    while pj_off == 'failed':
        po_status = pj.power.SetPowerOff(20)
        if po_status['error'] != 'NO_ERROR':
            pj_off = 'failed'
            sleep(1)
            logger.info("trying again to set PJ to poweroff")

        else: pj_off = "yay"
    logger.info("halting")
    subprocess.call(["sudo", "poweroff"])
else:
	logger.info("Exiting")

```

tvoverbeek, 2018-09-18:
What does your log say? Do you get retries?
One thing I would change is to move the line `subprocess.call(["sudo", "hwclock", "--hctosys"])`
after the check for /dev/i2c-1 and the pijuice instantiation (pj=...).
The OS also needs the I2C to communicate with the (simulated) RTC in the PiJuice MCU.
Also has the RTC itself been set correctly?

topshed, 2018-09-18:
> What does your log say? Do you get retries?

Yes, sometimes. I also get these errors when interrogating the PiJuice via the REPL

> One thing I would change is to move the line subprocess.call(["sudo", "hwclock", "--hctosys"])
> after the check for /dev/i2c-1 and the pijuice instantiation (pj=...).

Thanks. I'll try that. 

> Also has the RTC itself been set correctly?

Do you mean set to the correct time? Yes. If I check with pijuice_cli.py, it shows the correct time.

topshed, 2018-09-18:
I've moved the hwclock line as suggested. the PiJuice went through 3 booy/shutdown cycles then stopped.

Here's the edited log entries for the last successful  cycle:

[I 180918 17:45:22 pjdown4:39] 2018-09-18 17:45:22 -- Started
    
[I 180918 17:50:42 pjdown4:56] waiting (for network)
[I 180918 17:51:11 pjdown4:58] Taking measurements
[I 180918 17:51:11 pjdown4:67] Setting alarm
[I 180918 17:51:11 pjdown4:80] Cannot set alarm
[I 180918 17:51:12 pjdown4:80] Cannot set alarm
[I 180918 17:51:13 pjdown4:80] Cannot set alarm
[I 180918 17:51:14 pjdown4:80] Cannot set alarm
[I 180918 17:51:15 pjdown4:87] Alarm set for {'day': 'EVERY_DAY', 'minute': 57, 'hour': 'EVERY_HOUR', 'second': 0, 'month': 'EVERY_MONTH', 'year': 'EVERY_YEAR'}
[I 180918 17:51:15 pjdown4:93] Setting wakeup
[I 180918 17:51:15 pjdown4:105] Wakeup set
[I 180918 17:51:16 pjdown4:109] reading charge
[I 180918 17:51:47 pjdown4:151] Checking button
[I 180918 17:51:47 pjdown4:153] setting PJ to poweroff
[I 180918 17:51:48 pjdown4:160] trying again to set PJ to poweroff
[I 180918 17:51:49 pjdown4:160] trying again to set PJ to poweroff
[I 180918 17:51:50 pjdown4:160] trying again to set PJ to poweroff
[I 180918 17:51:51 pjdown4:160] trying again to set PJ to poweroff
[I 180918 17:51:51 pjdown4:163] halting

However the Pi failed to restart at 17:57

tvoverbeek, 2018-09-19:
@topshed I find it a bit suspect that there are exactly 4 retries for the setting alarm and the power off time delay. Since you have an internet connected RPi the wakeup enabled flag could be reset by the timedatectl daemon (at boot and about every 10-15 minutes).
This happens when you have the Pijuice id-eeprom at 0x50, which seems to be your case since you use `hwclock --hctosys`.
You can avoid this resetting of the wakeup flag by changing the id-eeprom address to 0x52 and load the rtc module via /boot/config.txt but specifying a ds1307 instead of a ds1339. The ds1307 does not have an alarm function, so the OS will not touch the wakeup flag.
Anyway I would move the setting of the wakeup flag to just before the power off.
Also as a debug you could include a `pj.rtcAlarm.GetControlStatus()` to check the status of the wakeup flag just before shutting down.
Still confused about your flaky I2C communication with the PiJuice

tvoverbeek, 2018-09-19:
@topshed One more thing, have you changed your i2c clock speed? Maybe trying to lower it might help.

topshed, 2018-09-23:
Changing the eeprom address appears to have corrected the problem. Thanks

closed 2018-09-23 by topshed