PiSupply/PiJuice #213: What are fault events?

richardquint, 2018-09-19:
I'm still coming to grip with my PiJuice and beginning Python.  While playing with the PiJuice Status functions I got a result of which I don't know the meaning.  GetStatus "is fault" is true, but on the Settings page Fault reports no fault. Is there any way to get more info on faults and what could they possibly be? I'm attaching screenshots of my script, its output and the Settings page.
[image]
[image]
[image]

tvoverbeek, 2018-09-19:
What is reported as Fault on the settings page comes from the output of pj.status.GetFaultStatus(),
but only 'battery profile invalid' and charging temperature faults are shown on the HAT pane.
For example my current output of pj.status.GetFaultStatus() is `{'data': {'button_power_off': True}, 'error': 'NO_ERROR'}`.
Definitely needs more documentation about the internals.

richardquint, 2018-09-19:
Okay, I get the same result for GetFaultStatus. See the screenshot. Where is that documented? Not on the https://github.com/PiSupply/PiJuice/tree/master/Software page, but there I do see that

"GetStatus() ... is_fault is True if there faults or fault events waiting to be read or False if there is no faults and no fault events." What does that mean in this case?  

Are there any other undocumented PiJuice functions?
[image]

tvoverbeek, 2018-09-19:
> Are there any other undocumented PiJuice functions?

Yes, just look at the available functions in pijuice.py (/usr/lib/python2.7/dist-packages/pijuice.py or
/usr/lib/python3.x/dist-packages/pijuice.py).
You can check the Firmware sources for exact details.
The 'isFault' flag in GetStatus() is a summary from all eventfaults ORed together:
From the firmware source:
```
powerOffBtnEventFlag;
forcedPowerOffFlag;
forcedVSysOutputOffFlag;
watchdogExpiredFlag;
((currentBatProfile == NULL) ? 0x20 : 0);
CHRGER_TS_FAULT_STATUS();
```
Only the last two are reported in the GUI/CLI