PiSupply/PiJuice #215: User Script for use with button

JW2606, 2018-09-22:
Hello, very new to this environment so apologies to all of you veterans.

I am exploring the new (for me and as I realise) and amazing world of Pi and have just purchased a new RPi PiJuice HAT.

The intention is on pressing button 2 that runs a simple script ( **4gdial.py** ) which simply runs a dial command from the command line prompt.

Having created a .py script I am able to run this from the command line with **python 4gdial.py**

But when I press the button, nothing happens (at least that I can see).

I have attached an image to hopefully display the relevant areas.

I've clearly missed something, but can't find any details in any of the forums for configuring these buttons and scripts.

Many thanks in advance.

J.

[image]

tvoverbeek, 2018-09-22:
A few things to fix:
On the Buttons tab you have assigned USER_FUNC_1  to SW2 PRESS. Note that SW2 is the outside one closest to the GPIO connector. The switches are in the order Sw1, SW3, SW2. You need to keep the switch pressed long enough to get PRESS recognized.
Your 4gdial.py has the executable bit set. Add the following line as the first line to 4gdial.py : `#!/usr/bin/python`. Then you can start the script with `/home/pi/4gdial.py`.
On the user-scripts tab change `python 4gdial.py` to `/home/pi/4gdial.py` (only if you changed the first line) or to `python /home/pi/4gdial.py`. User scripts are executed by the pijuice service which has /var/lib/pijuice as its working directory. Hence specify the full path of your user scripts.
One more thing. Check on the System Task tab if the 'System task enabled' is checked.
If the system task is not enabled no user functions are executed.

JW2606, 2018-09-22:
Thank you very much.  I'll try that now and revert back to you shortly.

J.

JW2606, 2018-09-22:
That worked. Thank you so much for your help.  System Task was enabled, but when I adapted the previous bash script I omitted the #!/usr/bin/python interpreter type at the top as you quite rightly corrected. 

I have read several different approaches on the best way to structure python scripts.  Some (as I have applied here) using the import os and others using subprocess call. 

This is the simple bash script I am looking to run;

`#!/bin/bash

sudo pppd call gprs #Connect to 4G Network

set -e
function contentsync {
        wget -O - -q -t 1 http://192.168.0.1?auth_token=token_here
}
trap contentsync EXIT

sudo killall pppd #Disconnect from 4G Network

echo "Script End"
echo ""`

The script is intended to connect using 4G synchronises the unit and then disconnect.

Would you suggest using the .py script to reference and therefor run this bash script or should I run this within the .py script.

Once again thank you for your help on this.

J.

tvoverbeek, 2018-09-22:
@JW2606 If you want to see/process the output from your script use the subprocess module to
run your bash script.
See the Python documentation for the subprocess module to see how to capture the output..