PiSupply/PiJuice #218: Wake Up Problem

mougz, 2018-09-24:
Hi, I'm currently testing a PiJuice for my raspberry Pi. What I want to do is to prevent power supply shortage of the RPi by running on battery when the power is not present (pretty simple I guess).
Our guess is that in a real environment, power shortage usually last little time,... but some times it can last longer. So our strategy is to shutdown ( or halt ) the RPi when the SoC is under a given threshold. Then, we want to wake up the PiJuice periodically to check if the power is back, in order to save energy. I've followed the instructions in the GitHub, but the RPi does not wake up after the given time when I shut it down( I modified the crontable btw) . So what I'm missing here ? Does shutting down or halting make a difference here ? I can send you part of my code if you want more precision Thanks

ChristopherRush, 2018-09-24:
Are you using the Python API? You should also be able to wakeup the PiJuice when power has been restored rather than checking periodically.

mougz, 2018-09-24:
Yes I am using the API  ! 
So what you're implying is that the Python code can run even if the RPi is shut down ? And if I check for the powerInput status regularly , I can wake it up when the power is present just by enabling the Wake Up ?

ChristopherRush, 2018-09-24:
No, what I'm implying is that you can configure the PiJuice in your code to wakeup when power has been restored:
```python
from pijuice import PiJuice 
pijuice = PiJuice(1, 0x14) 

configData = {'system_task': {'enabled': True}}

tl = 1
pijuice.power.SetWakeUpOnCharge(tl)
```

mougz, 2018-09-24:
Ok ! Thank you very much

closed 2018-09-24 by mougz