PiSupply/PiJuice #219: Shutdown and Wake Up with Spring Pin

mougz, 2018-09-25:
Hi,

I've managed to shutdown my RPi below a certain SoC and waking it up automatically when the power is back. Now I'm trying to do the same thing but when powering through power supply of the Pi. I've solder my spring pin and it seems to be well placed.

The problem is that  my RPi is shutting at the exact moment I cut the power and not below some percentage of battery. I try several times but it's always the same.However, the waking up on charge is working good.

What am I missing here ?

tvoverbeek, 2018-09-25:
@mougz 
Question: Do you have any action enabled on the 'No power' event (System Events tab of the GUI)?

mougz, 2018-09-25:
No !
However , I made some test and it was working when I was using GUI , but it did not work with my python code. I managed to find my mistake and now it work correctly.
Thanks for your answer anyway, i'm closing the issue

closed 2018-09-25 by mougz