PiSupply/PiJuice #221: State of Charge biased

mougz, 2018-09-28:
Hi,

I was wondering how State of charge (SoC) is computed. Indeed, I have huge difference of SoC (returned with function`pijuice.status.GetChargeLevel()` ) wether battery is charging or not.

When it's charging, SoC can rise to 90% even if SoC was 30% 2 min earlier without power supply.
It's kind of embarrassing because it is biasing my "WakeupOnCharge" function : if , for example, my trigger level is 30, the RPi will reboot even if the Soc is below 30 and i'm charging, because it will overestimate it.

continuum-kevin, 2018-11-07:
Wondering about this as well. If I leave the battery to charge even for several hours (presumably it should be fully charged at that point), charge percentage never rises above 99%. But then as soon as power is disconnected, it will drop to 50% or even 30% in just a couple minutes (only powering a Pi3b+, not even a display connected). Is this a problem only in how the charge is calculated, or is my battery not actually charging properly?

ChristopherRush, 2018-11-07:
The drop in percentage occurs because of the way the battery charge level is calculated. It calculates the charge based on the current load, as the load goes up the charge goes down. We are currently working on a better way to calculate the estimated charge level rather than in real-time and this will be done in a Firmware update in the future.