PiSupply/PiJuice #223: Battery profile

noahangel11, 2018-10-04:
I would like to use a different battery with my PiJuice, but apparently I may be setting up the profile incorrectly. It seems to show low battery in the GUI monitor fairly quickly. It is a large 4000mah battery and should last quite a while for my Pi 3B+ with a7”lcd display. Can you guys tell me how to set the custom profile for this battery? 
The battery info from the manufacturer is:
606090 battery
Standard Charging method ：0.5C （constant current）charge to 4.2V,then CV(constant voltage 4.2V)charge till charge current decline to ≤0.05C
4. Rated Capacity ： The capacity means the discharge capacity of the cell, which is measured with discharge current of 0.2C with 3.0V cut-off voltage after standard charge.
5. Charging time ：Standard Charging：3.0 hours(Ref.) ，Rapid charge： 2.0 hours(Ref.)
6. Max. charge current ：1.0C
Max. discharge current ：1.0C
Discharge cut-off voltage ：3.0V
7. Operating temperature ：Charging： -20℃~60℃ Discharging：-20℃~60℃
8. Storage temperature ： -10℃~ +45℃
9. Cycle life: 400 cycles,Stable performance with 100% in house testing 
10. Warranty: 12 Months 
11. Test condition：
Charge:0.5C to 4.2V
Discharge:0.2C to 3.0V
70% or more of 1st cycle capacity at 0.2C discharge of Operation 
Self-discharge ：Residual capacity >90%
After the standard charging, storied the cells under the condition for 30 days, then measured the capacity with 0.2C till 3.0V
12：Temperature characteristics ：
-20℃: ≥65%
23℃: 100%
60℃: ≥96%

noahangel11, 2018-10-10:
Anyone?

ChristopherRush, 2018-10-11:
How long does the battery last with your setup?

shawaj, 2018-12-17:
We are testing an experimental new firmware in #249 

Are you fairly comfortable with testing this kind of thing? If so, there is an ability with a new script to estimate some of the necessary settings for the battery profile. So it might help you if you are unsure.

Although please be very careful, as messing around with LiPos can be dangerous and we can't be held responsible for any issues caused with third party batteries.

noahangel11, 2018-12-17:
I will look into it thanks. Been busy with the holidays so the project has been sitting

closed 2019-09-15 by ChristopherRush