PiSupply/PiJuice #225: Powering PI B+, 7'' touch screen, and 2 PWM fans.

wilkinsmicawber, 2018-10-17:
I am looking to power the items mentioned in the title. Will I run into issues if I use a 10,000 maH battery? I plan to run the two .33A fans from the GPIO 5v connection.

tvoverbeek, 2018-10-17:
Will the 10,000 mAh battery be charged by the PiJuice? If it is a powerbank it will probably contain a multi-cell battery. PiJuice only supports charging single-cell battery.
When powering your setup directly from the powerbank (no PiJuice) it should work. Typically they can deliver 2.5A at 5V.
If it is a single-cell 10000mAh battery it should also work with the Pijuice.
No guarantee however, I have no experience with these large capacity single-cell batteries.

wilkinsmicawber, 2018-10-17:
You're right. The highest single-cell I can find is 5,000mAh. Thanks for the heads-up. I'd like to power everything from the PI-juice. I'll probably order everything and see how it goes.

wilkinsmicawber, 2018-10-17:
One critical issue that's come up is, I need to be able to guarantee the battery won't explode. I can't find any information on NTC temperature sensors being on any battery I can find, and the only LIPO batteries I can find with charge circuit protection are from Adafruit. If I can guarantee the Adafruit batteries won't explode they may suit the purpose, as 1.5 hours of up time is sufficient.

klint-k, 2018-10-18:
Have you figured out how to stack pijuice yet? that would help.

GizmoEV, 2019-02-05:
@tvoverbeek Why is this the case? "PiJuice only supports charging single-cell battery." If the cells are of the same chemistry and are put in parallel, the PiJuice will be unable to detect that it is two or more cells. It is impossible to electrically determine how many cells are in a parallel pack, even if the cells are of different capacity. They will all reach full at the same time and empty at the same time. I have tested this with LiFePO4 and used them in my Electric Vehicle starting over 9 years ago. Tesla has been using paralleled cells in their vehicles for years. I have dissected MacBook battery packs and they have three sets of paralleled cells. Inside the very cell that was supplied with my PiJuice is layer after layer of parallel cells. This is why this "restriction" doesn't make any sense to me.

I'm not trying to hijack this thread but it would @wilkinsmicawber make a decision about how he wants to power his system.

Thank you.

tvoverbeek, 2019-02-05:
@GizmoEV Read up over multi-cell LiPo charging and balanced charging.
To safely charge a multi-cell battery each cell should be charged individually.
Read e.g. https://learn.adafruit.com/multi-cell-lipo-charging/overview
Since the PiJuice does not have the circuitry for parallel charging it says use single cell batteries.
All the new PiJuice batteries (e.g. the 5000mAh and 12000mAh ones) are single cell.
For the Mac Apple makes sure the individual cells are precisely matched.

GizmoEV, 2019-02-06:
@tvoverbeek that article is on using cells in series and charging in parallel. That is quite different from always using them in parallel, which is what I am talking about. While the PiJuice batteries, like some pouch cells, are actually wrapped in a spiral as are cylindrical cells, not all pouch cells are spiral, some are actually alternating layers of anode, separator, cathode, separator, anode, etc with all of the anodes joined together and all the cathodes joined together. This makes multiple parallel cells. This is also how the 100Ah LiFePO4 cells I used in my car are made. I paralleled two cells for a 200Ah cell with the geometry required by the space. If there is no circuitry internal to the cell, then they will act exactly like a single cell when paralleled. Take the 12000mAh cell #9068135, for example. If two cells were paralleled at the terminals before the circuit board with NTC is attached then it will act exactly like a single 24000mAh cell. How do I know this? The theory supports it and my testing results confirm it as does years of field use.

As you can see in this cell pair from a MacBook, the two cell's terminals are welded together with the buss wire welded on top.
[image]
The other terminal is done the same way. The whole pack is in a 2P3S configuration. The monitoring and balance circuitry only measures and balances the series connected cells. For long term hands-off safety of the cells only series cells need to be monitored and balanced.

It is difficult to tell from the listing @wilkinsmicawber posted but it looks like that is a bare cell. If it is, any number of them could be paralleled without issue and the PiJuice circuitry would properly handle them fine.

shawaj, 2020-02-22:
@GizmoEV for obvious reasons, we don't really recommend people messing around with LiPo batteries. You obviously have done so successfully and have a huge amount of knowledge on the subject but it's not something that should really be attempted by everyone and so we can't really recommend it.

shawaj, 2020-12-05:
Closing this issue as no activity in quite some time. Feel free to reply if you disagree

closed 2020-12-05 by shawaj