PiSupply/PiJuice #228: Strange behaviour with wake up on charge

mougz, 2018-10-24:
Hi,

I've implemented the wake up on charge function on my PiJuice with trigger level set at 30%.
I'm using Python API. I've tried two different strategies : 
* 1) I'm using function "power.SetWakupOnCharge" with argument(30)
* 2) I'm directly changing the .json file (pijuice_config.JSON) with the json python parser. 
                    `data['system_task']['wakeup_on_charge']['enabled'] = True \n
                      data['system_task']['wakeup_on_charge']['trigger_level'] = 30`

When I'm testing solution 1), the Pi is waking up EVEN IF the PiJuice is still on battery and not charging.
When I'm testing the solution 2), I first check in the GUI if the changes are taken in account (and it's the case), but the Pi is not waking up when power is back.

What i'm missing here ?

ChristopherRush, 2018-10-24:
Is system task enabled?

mougz, 2018-10-24:
Yes, system task is enabled (I also modify the .JSON file `data['system_task']['enabled'] = True`)

Ok thank you for the tip, i will try it !

ChristopherRush, 2018-10-24:
Try reseting the configuration file by holding SW1+SW2 for 20 seconds. Here is a test script I use to test the wakeup on charge function using the API
```python 
import subprocess
from pijuice import PiJuice # Import pijuice module
pijuice = PiJuice(1, 0x14) # Instantiate PiJuice interface object

configData = {'system_task': {'enabled': True}}

tl = 1
pijuice.power.SetWakeUpOnCharge(tl)

pijuice.power.SetSystemPowerSwitch(0)
pijuice.power.SetPowerOff(60)
 
pijuice.status.SetLedBlink('D2', 3, [150, 0, 0], 200, [0, 100, 0], 200)
subprocess.call(["sudo", "halt"])
```

mougz, 2018-10-24:
Ok, your script was working well on my PiJuice.
I've managed to find my mistake : when I was powering the gateway off, I was using the "low charge" event with function SYS_FUNC_HALT_POW_OFF. I use SYS_FUNC_HALT instead and it works now !

Thanks for your help, I close the issue.

closed 2018-10-24 by mougz