PiSupply/PiJuice #229: pyjuice.status question

neil-nn11, 2018-10-24:
Hi, Can I please ask why the output of pyjuice.status toggles the order of 'error:' and 'data:' each time it is run. See screenshot of output. And can I stop it?
Kind regards

`#---tern-pi---------------------------
from pijuice import PiJuice
from datetime import datetime

pj = PiJuice(1, 0x14)
timestamp = datetime.utcnow()
BattVolts = pj.status.GetBatteryVoltage()
ChargeLevel = pj.status.GetChargeLevel()

print('------------------------------------------')
print(timestamp)
print(BattVolts)
print(ChargeLevel)
print('------------------------------------------')`

[image]

ChristopherRush, 2018-10-24:
Does this work:
BattVolts = pj.status.GetBatteryVoltage()[data]
ChargeLevel = pj.status.GetChargeLevel()[data]

neil-nn11, 2018-10-24:
Unfortunately not, I get NameError: name 'data' is not defined

ChristopherRush, 2018-10-24:
Sorry it should have been:
 ```ChargeLevel = pj.status.GetChargeLevel()["data"]```

neil-nn11, 2018-10-24:
That works perfectly thank you.

closed 2018-10-24 by ChristopherRush