PiSupply/PiJuice #230: SetWakeupEnabled doesn't work ?

Nyrknaqer, 2018-10-29:
Hi,
I don't understand why my raspberry doesn't wake up at schedule time.
I have a raspberry 2 B+ with raspbian-stretch-lite and I have installed 
the pogo pin.
I have set up the schedule through the PiJuice CLI.
I have shutdown my rpi and it boot correctly.
Where is the probleme ?
The probleme is that it won't boot after that if I shut it down again.
I have read that I must creat a python scrypt to set the wakeup alarm up 
again.
So I follow the instruction givin here : 
https://github.com/PiSupply/PiJuice/tree/master/Software#wakeup-alarm-menu
I have modified the script to understand what's wrong...

Here is my script :

```
#!/usr/bin/python
#
# This script is started at reboot by cron
# Since the start is very early in the boot sequence we wait for the 
i2c-1 device

import pijuice, sys, time, os

f = open('log.txt','w')

while not os.path.exists('/dev/i2c-1'):
time.sleep(0.1)
try:
   pj = pijuice.PiJuice(1, 0x14)
except:
   f.write('pj can not be created')
   sys.exit()

status = pj.rtcAlarm.SetWakeupEnabled(True)
s = str(('SetWakeupEnabled(True) : ',status))
f.write(s)
f.write('\n')

value = pj.rtcAlarm.GetControlStatus()
s = str(('GetControlStatus() : ', value))
f.write(s)
f.write('\n')
value = pj.rtcAlarm.GetAlarm()
s = str(('rtcAlarm.GetAlarm() : ',value))
f.write(s)
f.close()
```
and here the log.txt :
```
 ('SetWakeupEnabled(True) : ', {'error': 'NO_ERROR'})
 ('GetControlStatus() : ', {'data': {'alarm_flag': True, 'alarm_wakeup_enabled': True}, 'error': 'NO_ERROR'})
 ('rtcAlarm.GetAlarm() : ', {'data': {'minute_period': 5, 'second': 0,  'weekday': 'EVERY_DAY', 'hour': 'EVERY_HOUR'}, 'error': 'NO_ERROR'})
```
So it should boot, right ?
But it doesn't boot.
If I check in the Pijuice CLI it appears that the checkbox isn't 
checked...

Nyrknaqer, 2018-10-29:
I want to wake up my Pi every 5 minute to do a task then shutdown.
For the moment I'm just testing, so I shut it down manually.
So if I understand well, If an alarm is trigger when the Pi is running it will erase the wakeup flag.
The Pi should turn off before another wakeup alarm is trigger then, right ?
I'm going to shut it down automatically with this script :

```
#!/usr/bin/python

import os
from pijuice import PiJuice

pj=PiJuice(1, 0x14)
pj.power.SetPowerOff(30)
os.system("sudo shutdown now")

```
here my crontab:

```
# set wakeup alarm to reboot next time
@reboot /usr/bin/python /home/pi/python_scripts/wakeup_enable.py
# shutdown before a wakeup alarm from the pijuice
@reboot sleep 60 && /usr/bin/python /home/pi/python_scripts/power_off.py
```

Nyrknaqer, 2018-10-29:
It doesn't seem to work...
Any suggestions ?

tvoverbeek, 2018-10-29:
@Nyrknaqer How do you power your setup?
Since the pogo pin is mounted I assume you power the via the USB power input of the Pi, not the USB power input of the PiJuice. Correct?
In that case the SetPowerOff(30) has no effect since the Pi will still be powered.
Also has the PiJuice RTC the correct time?
The PiJuice RTC is simulated, so the PiJuice still needs power (albeit very little) to keep the PiJuice MCU running, so it can wakeup the Pi.
Other question: Is your Pi connected to the internet? If so the timesyncd when started will also clear the wakeup flag.

Nyrknaqer, 2018-10-29:
I power through the USB power input of the Pijuice. Do you mean that if I power through the Pijuice, I don't need to install the pogo pin ? (the documentation isn't clear)
The RTC time is fine.
My Pi has an internet connection.

tvoverbeek, 2018-10-29:
@Nyrknaqer When you power via the PiJuice you do not need the pogo pin.
Also set the 'Run pin: NOT INSTALLEDi' in the CLI.
To eliminate the possibility of timesyncd messing with the wakeup enabled flag, set the ID-EEPROM address to 0x52. This will disable the loading of the linux DS1339 driver (RTC with alarm function).
If you still want the syncing of the system time and RTC time from the internet when timesyncd starts, you can add the line `dtoverlay=i2c-rtc,ds1307` to /boot/config.txt. This will load the DS1307 driver (which has no alarm function) instead of the DS1339 driver. Of course, a reboot is needed to make these changes work.
Let me know how it goes.

Nyrknaqer, 2018-10-29:
That works great thank you

closed 2018-10-29 by Nyrknaqer