PiSupply/PiJuice #231: Battery drain while power is off...

LelandSindt, 2018-10-30:
I am using a PiJuice with a Rasberry Pi Zero W to allow the Raspberry Pi to come online when power is supplied to the Raspberry Pi's USB Port (I need power/data on the USB port for my application) and then shut down gracefully when power is removed. 

The following script is run via root's crontab every minute to detect power loss, set the RaspberryPi/PiJuice to power on when power is restored and then shut down... 
```
#!/usr/bin/python
import pijuice
import os
pj=pijuice.PiJuice(1, 0x14)
pj.power.SetWakeUpOnCharge(0)
if pj.status.GetStatus()['data']['powerInput5vIo'] == "NOT_PRESENT":
  os.system("sudo halt")
```

Overall the process is working as expected, the Pi comes on when power is available, and shuts down gracefully when power is removed. However, I am finding that the battery is draining in 8-10 hours (maybe longer) when the Pi is shut down. 

How long should I expect the battery to hold a charge when the Pi is turned off?

LelandSindt, 2018-10-30:
@ilyastrodubtsev 

No, sorry I mean discharge discharge time, specifically discharge time while the Raspberry Pi is off and the and the PiJuice is in stand-by.

tvoverbeek, 2018-10-30:
@LelandSindt  When your script shuts down the Pi it is powered by the 5V coming from the PiJuice.
After shutdown this power is still there (Red power led on Pi still on). The 5V will also continue to power your USB peripheral(s). Hence the 8-10 hours.
You need to add to your script to remove the 5V power from the PiJuice to the Pi:
```
#!/usr/bin/python
import pijuice
import os
pj=pijuice.PiJuice(1, 0x14)
pj.power.SetWakeUpOnCharge(0)
if pj.status.GetStatus()['data']['powerInput5vIo'] == "NOT_PRESENT":
  pj.power.SetPowerOff(20)  <=====
  os.system("sudo halt")
```
This will remove the 5V power after 20 seconds, leaving sufficient time for an orderly shutdown.
Then the PiJuice will go to a low power mode (slow weak blinking led on PiJuice).
It should be able to last more than a day in this state.

Can you confirm the Pi power LED stays on after shutdown?

LelandSindt, 2018-10-30:
@tvoverbeek 
When I remove power (without making your suggested changes) the Raspberry Pi shuts down as expected and D1 blinks green every ~5 seconds. 

Does it make any difference that I have the run pin/pogo pin installed and enabled?

LelandSindt, 2018-10-30:
[making the suggested change now]

LelandSindt, 2018-10-30:
With the `pj.power.SetPowerOff(20)` command I see the same ~5 second green blink from D1 once everything is shut down.

tvoverbeek, 2018-10-30:
@LelandSindt Waiting for your results.
You do need the pogo pin installed and enabled since you power through the Pi. Otherwise no way to boot the Pi.
(When powering via the PiJuice the boot is initiated by a short interruption of the 5V supplied by the PiJuice)

LelandSindt, 2018-10-30:
@tvoverbeek 
I have added `pj.power.SetSystemPowerSwitch(0)` and removed `pj.power.SetPowerOff(20)` (I would need to make some additional changes to my overall process to use a set/hard coded shutdown time)

The system will spend the majority of its time in standby today, I will review this evening when I return home.

LelandSindt, 2018-10-30:
.. fwiw, SystemPowerSwitch has been previously set to 0 via the pi_juicecli.py app.

tvoverbeek, 2018-10-30:
OK. Does the red Pi Power led go off after shut down?

LelandSindt, 2018-10-30:
The Raspberry Pi Zero [W] does not have a red (power) light. Only a yellow status/activity light. 

The Yellow activity light does turn off when the PiJuice/Rasberry Pi are powered down.

LelandSindt, 2018-10-31:
I put a volt meter on pins 2 and 6 (5v and Ground) While the PiJuice is on it reads ~5v. Once the PiJuie and Raspberry pi have powered down it reads ~.42v /most/ of the time. Occasionally it drops to 0v for a moment, then right back to ~.42v.

Is this the expected Behaviour?

See photo...
[image]

By the end of the day tomorrow I should have a charge state log that I can review and share.

LelandSindt, 2018-11-02:
I am going to go ahead a close this issue. When I have more data and time I may re-open... 
.... it would be nice to know if the ~.42v (see previous comment/photo) is expected.

closed 2018-11-02 by LelandSindt

tvoverbeek, 2018-11-02:
On my PiJuice connected to a 3B+ the voltage drops to 0.08V when the PiJuice 5V goes off and no
power connected to the PiJuice USB. Nothing else is connected to the 3B+.
The PiJuice battery on your photo does not look like my BP7X. A higher capacity one?

LelandSindt, 2018-11-02:
The battery is the BP7X that came stock with the PiJuice... (purchased via Amazon) 

If I remember/find time will will measure the voltage with a 3B+ and without a RPi see what it shows.

krishatech, 2022-04-28:
Same problem i face in pi4 when I press shutdown button it will shutting down and after 3 to 4 hours batteries  drain need help to understand,. 

 Two Batteries pack Sony 7c Li-ion 3000mAh and UPS HAT module for uninterrupted power supply on raspberry pi 4