PiSupply/PiJuice #232: HAT (PCB) assembled above PiJuice PCB

TomDidi, 2018-10-31:
Using PiJuice since around 5 month and I am very satisfied about its genius function and the stable work (form requirements).

There is only one question left regarding an additional PCB for a StarTech.com "USB to mSata Adapter" including mSata SSD, which shall be assembled on top of the PiJuice PCB .  Together with the PiJuice I bought the bigger sized case and now I want to use the housing for stacking Raspi3 in Groundfloor, PiJuice in 1st Floor and the mSata Adapter into the 2nd Floor.

I tried to assemble some standard standoffs (with male and female threads), so I removed the screws on the top side of PiJuice PCB and tried to assembly these standoffs. But unfortunately the length of the males thread is to short, due to the plastic fittings for the battery of the PiJuice.

So my simple question is, do you have any experience or knowledge where do I get such compatible standoffs with enough length on male's thread? 

Thanks in advance for your support.

Tom

KeithSloan, 2018-11-01:
Interested to know how this works out. I assume the Raspberry Pi HAT that you are talking about for attaching in addition to the PiJuice is https://www.startech.com/uk/HDD/Adapters/usb-sata-converter-raspberry-pi~PIB2S31

tvoverbeek, 2018-11-01:
Could  this (https://www.adafruit.com/product/3658) help? A collection of standoffs, screws etc. of various sizes.

TomDidi, 2018-11-01:
Just for giving more detailed information, sorry for missing them in my request.

1. for additional Pi HAT, which I want to assemble is the "USB to mSATA Converter for Raspberry Pi and Development Boards [https://www.startech.com/uk/HDD/Adapters/usb-msata-converter-raspberry-pi~PIB2MS1].

What I want to try, is to assemble this HAT above the PiJuice. And for this I have to remove the 4 screws from the PiJuice PCB and insert 4 standoffs (M/F) with a longer male thread. Cause the screws for standoffs from PiJuice are having also a longer male thread than the "normal" or standard screws for Pi HATs, due to the height of the plastic fitting for the battery, which are making the screws longer. 
But I need such standoffs with a longer male thread especially for use to add add HATs above PiJuice and a female thread on the opposite side to assemble the mSata HAT.

The "normal standoffs" are having a shorter male thread, so they will be not be fixed, they are "slack". And so I cannot assemble a HAT above the PiJuice. The only variance for the standoffs is their total length, but not the length of their threads.

2. the proposed selection from Adafruit is unfortunately not able to fit my demand for a longer male thread as explained before,sorry.

For making my problem more visible, please look air the attached picture, where you see the situation of Raspi3 with a PiJuice and the "big" housing from PiJuice.

[image]

Does anybody has an idea how to solve this problem?

tvoverbeek, 2018-11-01:
Other idea:
Use a gpio header extension (e.g. https://www.adafruit.com/product/1992) and mount
a perma-proto hat (e.g. https://www.adafruit.com/product/2310) on the gpio header.
This to get sufficient distance between the battery and the perma-proto hat.
Mount your usb-msata converter using the holes and standoffs on the opposite site of the gpio header.

ChristopherRush, 2019-09-15:
Suggestion above

closed 2019-09-15 by ChristopherRush

justin-cotarla, 2021-08-01:
I'm having the same issue, but with the [Dragino GPS + LoRa hat](https://www.dragino.com/products/lora/item/106-lora-gps-hat.html).

There has to be a better solution than just installing a dummy board between the Raspberry Pi and the next board. This not solve the problem, since the proto board itself is not mounted with standoffs, but is just "secured" via the gpio headers, with the rest dangling over the Pi. This is tantamount to just mounting the hat without standoffs directly via the gpio headers to begin with.

Going back to the original question, are there any non-standard standoffs that have a longer threaded section? Is there a standard way to deal with this issue?

tvoverbeek, 2021-08-02:
See e.g. https://www.adafruit.com/product/4685