PiSupply/PiJuice #234: V1.4 pijuice files invalid?

wboerhout, 2018-11-05:
Is  pijuice-base_1.4_all.deb invalid? `dpkg -i` fails, and the following shows why
pijuice_gui_all shows similar behaviour.

`pi@controller ~ $ sudo dpkg -c pijuice-base_1.4_all.deb`
`drwxr-xr-x root/root         0 2018-11-03 15:05 ./`
`drwxr-xr-x root/root         0 2018-11-03 15:05 ./etc/`
`drwxr-xr-x root/root         0 2018-11-03 15:05 ./etc/sudoers.d/`
`-r--r----- root/root       142 2018-10-13 16:04 ./etc/sudoers.d/020_pijuice-nopasswd`
`drwxr-xr-x root/root         0 2018-11-03 15:05 ./lib/`
`drwxr-xr-x root/root         0 2018-11-03 15:05 ./lib/systemd/`
`drwxr-xr-x root/root         0 2018-11-03 15:05 ./lib/systemd/system/`
`-rw-r--r-- root/root       261 2018-10-13 16:04 ./lib/systemd/system/pijuice.service`
`drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/`
`drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/bin/`
`-rwxr-xr-x root/root      9096 2018-10-13 16:04 ./usr/bin/pijuice_cli`
`-rwxr-xr-x root/root     87327 2018-11-01 14:23 ./usr/bin/pijuice_cli.py`
`-rwxr-xr-x root/root     11769 2018-11-03 15:05 ./usr/bin/pijuice_sys.py`
`-rwxr-xr-x root/root     20344 2018-03-16 21:47 ./usr/bin/pijuiceboot`
`dpkg-deb (subproces): kan het archiefelement van 'pijuice-base_1.4_all.deb' niet kopiëren naar de pijp van het decompressieprogramma: onverwacht einde van bestand of stroom`
`drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/lib/`
`drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/lib/python3.5/`
`drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/lib/python3.5/dist-packages/`
`drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/lib/python3.5/dist-packages/__pycache__/`
`-rw-r--r-- root/root     45303 2018-11-03 15:05 ./usr/lib/python3.5/dist-packages/__pycache__/pijuice.cpython-35.pyc`
`dpkg-deb (subproces): element van het archief wordt gedecomprimeerd: fout van lzma: gecomprimeerde gegevens zijn beschadigd`
`tar: Onverwacht einde-van-bestand in archief`
`tar: Error is not recoverable: exiting now`
`dpkg-deb: fout: subproces tar gaf een foutwaarde 2 terug`

tvoverbeek, 2018-11-05:
@wboerhout Just downloaded the pijuice-base_1.4_all.deb from github (to be sure to run the github version).
Here is my output:
```
pi@rpi-pijuice:~ $ sudo dpkg -c pijuice-base_1.4_all.deb
drwxr-xr-x root/root         0 2018-11-03 15:05 ./
drwxr-xr-x root/root         0 2018-11-03 15:05 ./etc/
drwxr-xr-x root/root         0 2018-11-03 15:05 ./etc/sudoers.d/
-r--r----- root/root       142 2018-10-13 16:04 ./etc/sudoers.d/020_pijuice-nopasswd
drwxr-xr-x root/root         0 2018-11-03 15:05 ./lib/
drwxr-xr-x root/root         0 2018-11-03 15:05 ./lib/systemd/
drwxr-xr-x root/root         0 2018-11-03 15:05 ./lib/systemd/system/
-rw-r--r-- root/root       261 2018-10-13 16:04 ./lib/systemd/system/pijuice.service
drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/
drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/bin/
-rwxr-xr-x root/root      9096 2018-10-13 16:04 ./usr/bin/pijuice_cli
-rwxr-xr-x root/root     87327 2018-11-01 14:23 ./usr/bin/pijuice_cli.py
-rwxr-xr-x root/root     11769 2018-11-03 15:05 ./usr/bin/pijuice_sys.py
-rwxr-xr-x root/root     20344 2018-03-16 21:47 ./usr/bin/pijuiceboot
drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/lib/
drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/lib/python3.5/
drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/lib/python3.5/dist-packages/
drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/lib/python3.5/dist-packages/__pycache__/
-rw-r--r-- root/root     45303 2018-11-03 15:05 ./usr/lib/python3.5/dist-packages/__pycache__/pijuice.cpython-35.pyc
-rw-r--r-- root/root     55580 2018-10-13 16:04 ./usr/lib/python3.5/dist-packages/pijuice.py
-rw-r--r-- root/root       255 2018-11-03 15:05 ./usr/lib/python3.5/dist-packages/pijuice_base-1.4.egg-info
drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/share/
drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/share/doc/
drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/share/doc/pijuice-base/
-rw-r--r-- root/root      1403 2018-11-03 15:05 ./usr/share/doc/pijuice-base/changelog.gz
drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/share/pijuice/
drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/share/pijuice/data/
drwxr-xr-x root/root         0 2018-11-03 15:05 ./usr/share/pijuice/data/firmware/
-rw-r--r-- root/root     70452 2018-09-23 21:08 ./usr/share/pijuice/data/firmware/PiJuice-V1.2_2018_05_02.elf.binary
pi@rpi-pijuice:~ $ 
```
No problems for me. Size of pijuice-base_1.4_all.deb should be 93254.
md5 sum:
```
pi@rpi-pijuice:~ $ md5sum pijuice-base_1.4_all.deb 
5007707ee2ebf31d45a29bf5e3c212d1  pijuice-base_1.4_all.deb
pi@rpi-pijuice:~ $ 
```
Please check if your download was corrupted.
Note the packages on github are not signed.

wboerhout, 2018-11-05:
Somehow, my Windows system failed to properly download using the "raw" file link. That has always worked. I now used the raw file link directly  from the pi, and that gave me an installable *.deb package.

Sorry for the confusion.

wboerhout, 2018-11-05:
However, after installing 1.4, my existing python scripts fail:
pi@controller ~ $ python led2off.py
Traceback (most recent call last):
  File "led2off.py", line 4, in <module>
    from pijuice import PiJuice
ImportError: No module named pijuice

*cat led2off.py*
#!/usr/bin/python
from pijuice import PiJuice
import time
import sys

pijuice = PiJuice(1,0x14)

led = 150

if len(sys.argv) > 1:
   led = int(sys.argv[1])

pijuice.status.SetLedState('D2', [led, 0, 0])
time.sleep(1)
pijuice.status.SetLedState('D2', [0, led, 0])
time.sleep(1)
pijuice.status.SetLedState('D2', [0, 0, led])
time.sleep(1)
pijuice.status.SetLedState('D2', [0, 0, 0])

wboerhout, 2018-11-05:
same with a script not written by me...

i@controller ~ $ python /usr/bin/pijuice_cli.py
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 15, in <module>
    from pijuice import PiJuice, pijuice_hard_functions, pijuice_sys_functions, pijuice_user_functions
ImportError: No module named pijuice

tvoverbeek, 2018-11-05:
@wboerhout Use python3 instead of python. 1.4 is python3 only.
Also start the cli with just pijuice_cli (no .py extension)
This starts the python3 interpreter running as non privileged user 'pijuice' executing pijuice_cli.py.
For a quick summary of changes see the changelog in /usr/share/doc/pijuice-base.
Docs still need updating before this is officially released.
In case of any more questions, please ask. Maybe you find some bugs ;-).
Al vast bedankt voor de feedback

wboerhout, 2018-11-05:
pi@controller ~ $ pijuice_cli
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 1779, in <module>
    lock_file = open(LOCK_FILE, 'w')
PermissionError: [Errno 13] Toegang geweigerd: '/tmp/pijuice_gui.lock'
pi@controller ~ $

tvoverbeek, 2018-11-05:
Delete the existing /tmp/pijuice_gui.lock (or reboot, which cleans /tmp).
Previously the lock file was owned by root, now by unprivileged user pijuice.
That is why you get a permission error.

wboerhout, 2018-11-05:
All I need is to change the I2C addresses, because I'm adding a ClusterHAT to the project.

wboerhout, 2018-11-05:
But the 0x14 address is built-in to pijuice_cli.py, right ?

wboerhout, 2018-11-05:
And I think I maneged to change the addresses. Do I need to change the EEPROM address to 52 as well?
`pi@controller ~ $ i2cdetect -y 1`
`     0   1   2   3   4   5   6   7   8   9   a   b   c   d   e   f`
`00:          -- -- -- -- -- -- -- -- -- -- -- -- --`
`10: -- -- -- -- -- -- 16 -- -- -- -- -- -- -- -- --`
`20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --`
`30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --`
`40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --`
`50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --`
`60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --`
`70: 70 -- -- -- -- -- -- --`

shawaj, 2018-12-17:
@wboerhout how did you get on? is everything you need working now?

wboerhout, 2018-12-18:
Not yet, but that has to do with logistics. My daughter in Cornwall is the
shipping intermediate. She will bring the device to me (in NL) this Sunday.
After Christmas, I will let you know. And she will send the original board
back after her return to UK.

Wilm

wboerhout, 2018-12-28:
> <notifications@github.com <mailto:notifications@github.com>>

I'm all good now.

Wilm

shawaj, 2018-12-28:
Perfect, I'll close this issue now then

closed 2018-12-28 by shawaj