PiSupply/PiJuice #235: IO on P3 question

neil-nn11, 2018-11-06:
Could some kind person please help me with how I use the two IO pins on P3. The documentation states;

SetIoDigitalOutput(pin, value) Sets state at IO pin configured as digital output. Arguments: pin: IO pin designator, 1 for IO1, 2 for IO2. value: output state to set, 0 for low output state, 1 for high. Example:
print pijuice.status.SetIoDigitalOutput(1, 1)

I have tried this but it does not seem to do anything so this leads to my question.... does '1 for high' mean the pin goes high as in 3.3v? (which is the desired thing I'm after).

tvoverbeek, 2018-11-06:
@neil-nn11 This assumes you have set the io configuration for the desired pin to output.
Use the gui or cli to set the output mode and pull up/down if needed.
See the IO description in the software README (either for the gui or for the cli).
The configuration you set gets stored in non-volatile memory, so it is preserved over PiJuice restarts.
Then you can use pijuice.status.SetIoDigitalOutput() to control the pin.

neil-nn11, 2018-11-07:
Ah I see. Thanks.

rburgul, 2019-02-15:
Is high 3.3v or 5v? I’m hoping to use these pins with the HC-SR04 ultrasonic sensor. 
Thanks.

tvoverbeek, 2019-02-15:
@rburgul According to the PiJuice schematic (https://github.com/PiSupply/PiJuice/blob/master/Documentation/PiJuice_Schematic.PDF) IO1 and IO2 are connected to PA7 and PA8 respectively.
The STM32F0 datasheet says PA7 is TTa(3.3 V tolerant I/O directly connected to ADC)
PA8 is FT(5 V tolerant I/O)
So you could use IO1 for the trigger pulse and IO2 for the echo input.
Note there also 3.3V versions of the echo sensor. See e.g. https://www.adafruit.com/product/4007

rburgul, 2019-02-20:
Thanks very much. I connected it directly as above. But struggled to make it work. On testing voltages, I could see io1 changing to 3.3v but never got an echo. I might have to try connecting back to the gpios.

tvoverbeek, 2019-02-20:
@rburgul Since programming the PiJuice IOs is via the Python API it will be too slow to pick up the echo after you send the pulse on the HC-SR04.
You will be better off using the GPIOs (with a 3.3/5V converter)

rburgul, 2019-02-20:
Yes. Agree. I thought that may be the problem. Before I re-wire, is there any way of accessing io1 and io2 directly without using the API? This issue renders these pins unusable for fast devices. A pijuice node-red module might be an idea?
BTW you are very responsive and generous with your time on these forums. Thanks you so very much!

tvoverbeek, 2019-02-20:
@rburgul You could, but you still need to communicate via I2C to the PiJuice. For the ultrasonic distance sensor I believe this will still be too slow.

rburgul, 2019-02-20:
Ah. Yes of course. 
Thanks again.

closed 2019-09-15 by ChristopherRush