PiSupply/PiJuice #238: Below freezing causes corruption

hp1337, 2018-11-12:
I am using the pijuice outdoors in Canada. It sometimes goes below zero Celsius. I have noticed very erratic behavior when this happens. Random LEDs will start blinking. Phantom button presses will be detected. And eventually complete freezing of the PiJuice microcontroller. This is without the Raspberry Pi powered. The only solution is to remove the battery to reset the PiJuice

This only happens if battery powered. If I plug into a wall outlet while below freezing  then everything works.

I think the code that processes battery temperatures is mishandling negative numbers. I even changed the "Temperature Sense" setting to "Not Used" and it still fails.

How can I get around this issue? Seems silly that below freezing is not supported. Lithium ion should be able to provide reasonable current evem below minus 10 Celsius.

EricS61, 2018-11-12:
OK, I'm not the only one who's Pi shut down due to below freezing temps. I leave mine out in the vehicle, as I use that particular Pi as a DMR MMDVM/Pi-Star interface for my mobile radio. Guess will have to bring it indoors for the time being as temps will get down to -20 to -50 Fahrenheit for a week or two come January.

shawaj, 2018-11-12:
This is from our engineer @mmilann ...

All components specified for PiJuice are in Industrial temperature range -40C - +85C except rgb LEDS (-20C), but LEDs have somewhat self heating and probably can go below.

Attached are component temperature range specs. Most below zero temperature sensitive is battery that can decrease capacity. I did some tests over winter on first prototype below -10C and it worked with battery. 

Yes there is possibility that this problem lays in firmware because of lack of cold test conditions during debugging in development.

shawaj, 2018-11-12:
Also more from @ChristopherRush and @mmilann 

If users are charging their battery at below zero temperature then it could potentially cause Lithium plating, damaging the battery permanently.

Yes and there is limitation in firmware so when battery temp drops below "Cold temperature" settings which is set to 1C for BP7X charging will be disabled preventing damage to battery.
This is according to JEITA compliance: http://www.ti.com/lit/an/slyt365/slyt365.pdf for safety charging. 

This problem can be caused by firmware error because battery voltage drops as temperature goes down and if it drops below 3.3V there can be errors in ADC measurement.

Also...
[image]

hp1337, 2018-11-12:
> If users are charging their battery at below zero temperature then it could potentially cause Lithium plating, damaging the battery permanently.

My issue is with DISCHARGING at freezing temperatures. I know it is VERY BAD to charge a Lithium Ion battery below zero.

Please consider the following scenarios I have observed:

1. Below 0 deg Celsius, Battery Removed, Plugged into wall: PiJuice Functions Normally
2. Below 0 deg Celsius, Battery in place, NO CHARGING: PiJuice becomes erratic, System voltage is regulated well! I can draw current from battery without issue!

Scenario 2 tells me that the Battery, and voltage regulators are working great. This points to the FIRMWARE misbehaving.

shawaj, 2018-11-12:
Thanks @hp1337 we are investigating further

shawaj, 2018-12-19:
@hp1337 it does not look like a firmware issue as far as we can see.

From our engineer, Milan: 
```
I did test below zero at around  -14C, -12C.

BP7X 1820 labelled with pijuice, internal impedance significantly rises to around 1.5 Ohm at -12C. That means
there is voltage drop across battery, and when rpi draw 0.6 amps voltage can drop below minimum (3V) causing 5V
regulator to shutdown. So this extreme low temperatures lowers maximum current that can be drawn from battery.

Recommended operating range with Lipo is 0C-60C, below zero users need to use kind of thermally isolated case or container.
```

tvoverbeek, 2019-01-17:
@hp1337 Looks like you were right.
Here some info from @mmilann : 
```
Have discovered that below zero there is temperature measurement firmware bug,
it report over 250C when cool battery below zero,
```
Smells like an unsigned byte vs signed byte issue.
Should be fixed in the latest Beta 1.3 firmware from Jan 15, but I have not verified this.

shawaj, 2020-02-22:
@hp1337 @EricS61 is the latest firmware working now for you?

EricS61, 2020-02-23:
Aaron,

 I've found out that charging LiPo's aren't the best thing to do in extremely cold weather, so my PiJuice setups have remained indoors till the evening/daytime temps stay above freezing. I don't even plug my cellphone (LiPo battery) into it's charger when the temp inside the cabin of the vehicle is below freezing anymore. But, will update firmware & software when my task list dwindles down a bit!  Thanks for asking, and you can close this if majority agrees.

v/r,

 Eric

shawaj, 2020-12-05:
Closing as no activity in over a year

closed 2020-12-05 by shawaj

Jamos1988, 2021-02-25:
Just wanted to know, is the status of the firmware still the same around this subject?

tvoverbeek, 2021-02-25:
Yes, no charging below freezing (0 deg. C)

Jamos1988, 2021-02-25:
Thank you, I do wonder how users now mitigate with this issue in "off-grid" projects. I've just read this article: https://rogershobbycenter.com/lipoguide/ and I understand that it is better to just store the battery elsewhere in case of freezing temperatures, for longer durability. But, for safety reasons I am wondering what would happen if I just kept the system running when temperatures are dropping. For now I am thinking of sending notifications / alarms when this happens, so the system will shut down or stop charging? Just thinking out load, wondering how other users see this? Some background, I am planning to use this system on a boat, which sometimes will be unattended for longer periods of time.