PiSupply/PiJuice #239: System Task options not working as expected

DivingIvan, 2018-11-14:
I am finding some very strange behaviour with the 'System Task' options. I am using software version 1.4 and firmware version 1.2.

I am trying to get the 'Software Halt Power Off' to switch power off after a set time (30 seconds). However, I cannot get this option to work.

What I am also finding, however, is that the system will automatically power on again after 2 minutes; even though I have the 'watchdog' option set to disabled. So, every time I shutdown the system, it will automatically start again after 2 minutes.

ilyastrodubtsev, 2018-11-16:
@DivingIvan Hello. I can't reproduce your problem. My raspberry don't restart at the expiration of 1,5 hours even with system task "Software Halt Power Off". Please describe your problem in more detail.

DivingIvan, 2018-11-16:
Hi, thanks for the response. I now suspect that the problem is more deep seated, as I am finding a lot of other functions do not work either; for example, the Pi does not shut down when it reaches the low battery threshold that has been set, and will not restart after power is reinstated once the battery reaches the charge threshold that has been set.

I'm going to do a clean install and see if I still get problems.

DivingIvan, 2018-11-16:
Right, I've started from a clean Raspbian install (latest - 2018-11-13). I've downloaded the 2 PiJuice Debian packages for version 1.4 (pijuice-base_1.3.3_all.deb and pijuice-gui_1.4_all.deb), and run commands:-

sudo dpkg -i ./pijuice-base_1.1-1_all.deb

and

sudo dpkg -i ./pijuice-gui_1.1-1_all.deb

Following reboot, I've set the 'Software Halt Power Off' checkbox, and given a delay period of 30 seconds. However, when I shutdown the Pi, the power does not get switched off after the 30 second delay period.

I then tried setting the 'watchdog' checkbox, and gave an expiry period of 1 minute. When I shutdown the Pi, the Pi did NOT restart after 1 minute.

However, I then rebooted the Pi, and this time unset the 'watchdog' checkbox. This time when I shutdown the Pi, the Pi DID restart after 1 minute.

So, none of this seems to be working as I would expect. The watchdog seems to do completely the opposite of what is configured, and the 'Software Halt Power Off' doesn't switch off the power to the board no matter how it is configured.

ChristopherRush, 2018-11-16:
How are you powering your board?

DivingIvan, 2018-11-16:
Power supply into PiJuice board micro USB socket.

ChristopherRush, 2018-11-16:
How are you configuring the PiJuice? Are you using the GUI, CLI or API?

DivingIvan, 2018-11-19:
I'm using the GUI to configure the PiJuice, and from what I can tell, changes to the configuration are being written to the /var/lib/pijuice/pijuice_config.JSON file.

I'm beginning to wonder whether it's the PiJuice firmware that may be the issue, since this is what will ultimately handle the watchdog and 'software halt power off' functionality.

However, as mentioned previously, I am also finding that the PiJuice does not shutdown the Pi when it reaches it's critical low battery charge (I've got this set at 5%); and this, I'm assuming, would be handled by the PiJuice service.

ilyastrodubtsev, 2018-11-19:
Tell me, after which you noticed the inoperability of system tasks.

DivingIvan, 2018-11-19:
The problems started after I installed the PiJuice version 1.4 package. Everything seemed to work fine using version 1,3.

However, I specifically wanted the 'software halt power off' functionality, which wasn't available in version 1.3; which is why I installed version 1.4.

ilyastrodubtsev, 2018-11-19:
Do not work only tasks "minimum charge" and "Software Halt Power Off"?  Are other functions performed correctly?

DivingIvan, 2018-11-19:
I've not tested everything, but I'm finding a lot of functions that do not seem to work as expected.

The 'software halt power off' doesn't power off the Pi, the 'minimum charge' function does not shutdown the Pi at the defined battery threshold, and the watchdog switches the Pi back on when the watchdog is disabled, but doesn't switch it on when it IS enabled.

DivingIvan, 2018-11-20:
Ok, I guess that would explain why the power off function isn't working.
Do you know when the 1.4 release is likely to be officially available?
Meantime, I'll reinstall the 1.3 release, as I don't think I had any problems with that version.
Many thanks.

DivingIvan, 2018-11-20:
So, the plot thickens.
I've removed release 1.4. and reinstalled release 1.3.
I now find that the PiJuice will shut down the Pi at the low battery threshold, and it then automatically powers off the Pi 1 minute after the shutdown (which is pretty much what I was trying to achieve with the 'software halt power off').
However, the watchdog function seems to power the Pi back on again (2 minutes after the shutdown) when I have the watchdog disabled. To stop the watchdog from restarting the Pi, I have to set it to enabled.

tvoverbeek, 2018-11-20:
@DivingIvan I can imagine a scenario to explain what you are seeing.

The watchdog is enabled at start of the pijuice service (normally at boot) when it is enabled in settings  with a non zero timeout in minutes.
The PiJuice watchdog is reset every time it receives a command via I2C. Normally the service is polling the PiJuice wit I2C every 2 seconds. On the desktop there is additional I2C polling by the battery charge icon.
When the Pi is shutdown by the PiJuice (e.g. by SYS_FUNC_HALT_POW_OFF on low charge) and the PiJuice watchdog is enabled in settings, it is disabled just before the PiJuice sends the shutdown command to the Pi.

Now imagine the following scenario:
- Pi is booted with watchdog enabled and timeout set to 2 minutes -> watchdog is active.
- The watchdog is disabled in the GUI. Watchdog is only disabled in settings, but remains active.
- Pi is shutdown due to low charge with SYS_HALT_POW_OFF. Since watchdog is now disabled in settings the disable watchdog command is not sent.
- 2 minutes later (1 minute after the power led goes off) the Pi restarts.

If you now wait till it shuts down again due to low charge or via a button press coupled to SYS_HALT_POW_OFF it should not reboot again, since the watchdog was not enabled at boot.

In 1.4 when you send the shutdown command yourself (either from terminal, desktop menu, or a program) the watchdog will also be disabled at shutdown as when using SYS_FUNC_HAT_POW_OFF.

DivingIvan, 2018-11-21:
@tvoverbeek - I'm still wondering whether there might be a slight flaw in the logic of the PiJuice service with regards to the watchdog.

I've tried the following a few times now, and it consistently fails to operate as I would expect. The scenario is as follows:-

1 - The 'system task enabled' flag is set true on the system task tab of the gui, and the 'watchdog' flag is set to true with an expire period of 2 minutes.

2 - The Pi is rebooted.

3 - The 'watchdog' flag is now set to false on the gui, but the expire period is left unchanged (it's greyed out).

4 - I shut down the Pi, and it restarts after 2 minutes.

5 - I shut down the Pi again, but it still restarts after 2 minutes.

etc, etc. Every time I now shut down the Pi, it automatically restarts after 2 minutes.

My thinking is that, because the 'watchdog'  flag is set to false, the signal is never sent to the PiJuice to tell it to turn off the watchdog functionality.

tvoverbeek, 2018-11-21:
@DivingIvan Thanks for testing.
I can reproduce what you are seeing.
We have a similar problem when the pijuice service is started.
At the end of the previous scenario the watchdog was disabled in settings, but still active in the PiJuice. When booting and the service starts it only sets the watchdog when enabled in settings with a period > 0. It does *not* disable it when it is set to disabled in settings. Hence it stays where it is, e.g. enabled with a timeout of 2 minutes.

I'll try to add a few changes to the code, so when the setting is changed in GUI/CLI the watchdog setting in the PiJuice is also changed. Also at service start, the watchdog will be set to disabled when disabled in settings.
I will let you know when done, so you can test it.

Again thanks for chasing this. Appreciate your tenacity in testing this.

tvoverbeek, 2018-11-21:
pijuice-base 1.4 package in Software/Install updated with Watchdog fix.
@DivingIvan Can you test the 1.e packages now?
Please do the following:
```
- sudo apt-get purge pijuice-gui pijuice-base
- Get the pijuice-base_1.4_all.deb and pijuice-gui_1.4_all.deb from Github
- sudo dpkg -i pijuice-base_1.4_all.deb
- sudo dpkg -i pijuice-gui_1.4_all.deb
- Reboot just to be sure files in /tmp are cleared.
```
Note in 1.4 to start the GUI/CLI use `/usr/bin/pijuice_gui` and `/usr/bin/pijice_cli` (No .py extension)
This is because the GUI and CLI now run not as user pi but as user pijuice.
Also /var/lib/pijuice/pijuice_config.JSON is now owned by pijuice and not world readable.
Use a root shell (`sudo -i`) to read / change the JSON file manually.

Let me know how it goes.

PS I assist PiSupply with the code as volunteer, but PiSupply will have to make the packages available in Raspbian.

closed 2018-11-22 by DivingIvan