PiSupply/PiJuice #240: Wake Up on Charge inconsistent behaviour

mougz, 2018-11-15:
Hi,

We have setup our pijuice to wake up when on charge and above 30% of SoC (trigger level). The pijuice is also configured to halt when battery is below 30%(minimum charge + LOW_CHARGE event : SYS_FUNC_HALT)

This behaviour works perfectly when we unload the battery to 30% : the Pi halts and when we put back the charge on the Pi wake up.
However, we want our Pi to wake up whenever the PJ is charging, in order to imitate the behaviour of a Raspberry Pi without PJ ( ie the Pi must turn on when charge is on). So we have plugged the power connector off, then halted the system and put the charge back on. But the Pi is not waking up in this case.
What are we missing ? 

PS : The battery  is always charging from PiJuice USB power connector.

mougz, 2018-11-15:
Yes that's right !

We follow your instructions  and used SYS_FUNC_HALT_POW_OFF. We also set trigger level to 0%. The red led turn off after 60 sec but when we put the power input back, the PI does not wake up.

mougz, 2018-11-15:
What is weird is that wake up **works perfectly** when we use SYS_FUNC_HALT and PJ is halted by the minimum charge event.

But when we used SYS_FUNC_HALT_POW_OFF (associated to a switch) , it does not wake up when power is back.

mougz, 2018-11-15:
We just the want the Pi to automatically switch on when power is present.

tvoverbeek, 2018-11-15:
@mougz Is your Pi connected to the Internet? What is the id-eeprom address?

mougz, 2018-11-15:
Yes we are connected to Internet. the ID eeprom adress is 50.

tvoverbeek, 2018-11-15:
@mougz The internet connection and id-eeprom address do not influence the wakeup-on-charge.
I was confusing it with the periodic wakeup by alarm.
Here is how the wakeup on charge is supposed to work.
When you change/set the wakeup on charge in the gui/cli only the JSON settings file is updated. The setting is not sent to the PiJuice. When the SYS_FUNC_HALT_POW_OFF is executed by the system task the setting is sent to the PiJuice just before the shutdown.
Things to check:
- Is the pijuice service active? (The service runs the system task pijuice_sys.py). I suppose it is
  since otherwise the shutdown at 30% will not happen.
- Is the wakeup-on-charge setting in the JSON file? (If not, it will not work)

mougz, 2018-11-16:
Hi @tvoverbeek 

ALL the settings are in the JSON file. (wake up on charge enabled, system task enabled, trigger level at 0,...). The pijuice service seems very active to me , as the shutdown happens at 30 % as you said. For testing purpose ( because waiting for the battery to be at 30% can be time consuming), I've set the minimum charge at 90%, with low charge event set at "SYS_FUNC_HALT_POW_OFF". The PiJuice behave as expected in this case ( system halts, then the red led on the Pi turns white after 60s, and the system wakes up well when I put back the power input )

I've tried this morning to do the same thing but instead of halting and powering the PI off with low charge event, I want to call SYS_FUNC_HALT_POW_OFF with a button. And it does not work. I've set it to different switch with different event (single press, long press), I'm sure I've applied the settings , but when I push the related switch, nothing happens. 
Note that we already tried this yesterday and it worked **very well**. The wake up did not work, but the switch press sure did. 
I cannot understand why there is a difference in behaviour when we are calling SYS_FUNC_HALT_POW_OFF with switch and when this function is called by an event. It's really really strange.

Again, what i found disturbing is this **lack of consistency** in the PJ behaviour. Some feature works very well at one moment but can fail you one day after. Some error can happen but its nearly impossible to reproduce it in order to fix it.

drv2e30, 2018-12-31:
Could this be an issue with not using the pogo pin?

Would it be best to just set an alarm on a schedule to turn on the unit?

closed 2019-09-15 by ChristopherRush

drv2e30, 2019-09-20:
Hey,

Just so you know I ended up putting a diode in between.

Thanks,

-Sean