PiSupply/PiJuice #242: Cant download PJ 1.4

Danielcor17, 2018-11-23:
Tired apt-get
Tried .deb
tried git clone

Cannot successfully install PJ 1.4

As well I suspect this is the reason why I do not see an option for Software Halt Power Off under System Tasks. Is this why or is that another issue?

tvoverbeek, 2018-11-23:
@Danielcor17 
apt-get obviously does not work since the 1.4 deb packages in github have not been submitted to the Raspbian archive yet.
If you did 'git clone' you have downloaded the .deb packages. You'll find them in PiJuice/Software/Install.
Here is how to install them:
```
sudo apt-get purge pijuice-gui
sudo apt-get purge pijuice-base
sudo dpkg -i <path to>pijuice-base_1.4_all.deb
sudo dpkg -i <path to>pijuice-gui_1.4_all.dev
```
Reboot after this, just to be sure to start clean.

Danielcor17, 2018-12-13:
Okay that worked. Very grateful for the new software halt power off option. Thanks for the support!

closed 2018-12-13 by Danielcor17