PiSupply/PiJuice #243: PiJuice HAT CLI - Crash

dgrimani01, 2018-11-23:
Pre-requirement: this test is performed with firmware 'PiJuice-V1.2_2018_03_28.elf.binary' and 'PiJuice-V1.2_2018_05_02.elf.binary' with same results.
 
1) I run the pijuice_cli.py script.
2) I set the Wakeup Alarm function for first time by PiJuice HAT CLI --> Wakeup Alarm function.
2) I execute the shutdown and i wait the automatic reboot. Thats work all OK. My raspberry startup on time.
3) The crash of pijuice_cli.py script appear after automatic reboot when i run one more this script and i go to PiJuice HAT CLI --> Wakeup Alarm.
4) The problem disappear if i re-flash the firmware with command: ./pijuiceboot 14 PiJuice-V1.2_2018_XX_YY.elf.binary.

 
The trace of crash as follow:

**Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 1229, in <module>
    urwid.MainLoop(top, palette=[('reversed', 'standout', '')]).run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 719, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python2.7/dist-packages/urwid/raw_display.py", line 393, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python2.7/dist-packages/urwid/raw_display.py", line 493, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 403, in _update
    self.process_input(keys)
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 503, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python2.7/dist-packages/urwid/container.py", line 592, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python2.7/dist-packages/urwid/decoration.py", line 621, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python2.7/dist-packages/urwid/decoration.py", line 621, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python2.7/dist-packages/urwid/listbox.py", line 987, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python2.7/dist-packages/urwid/wimp.py", line 535, in keypress
    self._emit('click')
  File "/usr/lib/python2.7/dist-packages/urwid/widget.py", line 464, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python2.7/dist-packages/urwid/signals.py", line 264, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python2.7/dist-packages/urwid/signals.py", line 294, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1191, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 983, in __init__
    self.device_time = self._get_device_time()
  File "/usr/bin/pijuice_cli.py", line 1052, in _get_device_time
    str(t['second']).rjust(2, '0')))
  File "/usr/bin/pijuice_cli.py", line 1058, in _update_status
    self.status_text.set_text("Status: " + self.status)
AttributeError: 'WakeupAlarmTab' object has no attribute 'status_text'**

dgrimani01, 2018-11-26:
This issue as been resolved. I have downloaded the latest version of PiJuice version on gitHub

closed 2018-11-26 by dgrimani01

tvoverbeek, 2018-11-26:
@dgrimani01 Glad you got it resolved. If you detect any other problem with pijuice-cli let us know,